wled/WLED #1408: Board constantly going offline until reboot

RandomPotato893, 2020-12-01:
**Describe the bug**
My wemos d1 mini (I think it's a knockoff) will connect to wifi with the firmware just fine but then it goes offline. It stays offline until I reboot it. Then it goes online for another 1 minute and then it dies again. I have already seen some people with a similar issue which is #424 I seem to have tried everything on that issue list (including building my own firmware and doing the ping fix) with no success. I have tried to compile different versions and older versions and nothing seems to work. I have my esp in a box outside running some fancy Christmas lights but it does the same thing when it is in my room right next to my router. I just wanted to see if anyone else has this problem and if anyone can help me out. 
**To Reproduce**
I connect to the access point, I get it on my network (I have an amplifi instant wifi network if it matters) Then it boots up and goes offline after about 30s to 1 min. Then if I power-cycle the board. It goes back online and repeats the process.

**Expected behavior**
I expect it to boot up and stay online

**WLED version**
 - Board: [D1 mini]
 - Version [A bunch but I have been working with 10.0.2 and 0.11.0 (the newest one)]
 - Format [I have used bin files and compiled by myself using platformio and it makes no difference.]

Thanks to anyone with any ideas!

Aircoookie, 2020-12-01:
Hi!
I am not sure if this can help you, but an Ubiquiti user found a solution to his WiFi problem here: https://github.com/Aircoookie/WLED/issues/1293#issuecomment-717929824

Since Amplifi seems to be a sub-brand of Ubiquiti, maybe this applies to you as well, I don't know though 👍

RandomPotato893, 2020-12-01:
Thanks, sorry for the late reply I was at school and I didn't get GitHub notifications for some reason. I will try that right away. I appreciate you taking the time out of your day to help me. This is some amazing software

RandomPotato893, 2020-12-01:
Sadly, I don't think this works with my system. Amplifi is mainly controlled through an app and it is different from the normal systems. I can't find that setting and I don't see anything about it online for my setup. Thanks for the idea and I hope that I can find a solution soon because I really like the led effects in this software. Furthermore, I have had a similar setup (exact same board) on my network before (this one was working perfectly) and then I reflashed it with a different software version and it broke. Thanks for the idea though!

RandomPotato893, 2020-12-02:
Does anyone else have any other ideas?

ulte, 2020-12-03:
@RandomPotato893 i had this in the previous version before 11. I resolved it by using one of the 2 usermods that essentially self pings the ip address as a “keep alive” of sorts. That worked perfectly for me. Prior to this I noticed if I used my laptop to keep a constant ping going to the wled ip it would never go offline. Obviously the usermod is a more practical solution. Have you tried them?

RandomPotato893, 2020-12-03:
Sorry to ask an obvious question but what does that entail? I'm kinda new to this

RandomPotato893, 2020-12-03:
Is it the one by @gegu Because I already tried that and it didn't seem to do anything. I could've done it wrong though.

ulte, 2020-12-03:
It was that one but make sure you follow his instructions. I had to recheck my work a few times to ensure I correctly modified the usermods_list.cpp correctly. Also if you did it right you’ll see # of pings in the info window.

RandomPotato893, 2020-12-03:
Ok I thought it was just usermods.cpp

RandomPotato893, 2020-12-03:
He doesn't have a file to download right? I have just been making one and pasting the text in.

ulte, 2020-12-03:
you do have to also move a file into the wled00 directory. See his instructions

RandomPotato893, 2020-12-03:
The only thing I see is the installation. I don't see any specific instructions. Sry again if i'm being slow

RandomPotato893, 2020-12-03:
All I see is this:
[image]

RandomPotato893, 2020-12-03:
Sorry again if I'm being slow, perhaps I could have an exact link?

ulte, 2020-12-03:
ahh I see. Yea its the other usermod: https://github.com/Aircoookie/WLED/tree/master/usermods/Fix_unreachable_netservices_v2

RandomPotato893, 2020-12-03:
Ohhh Thanks!

RandomPotato893, 2020-12-03:
So I have been at this for a few hours with no success. When I tried it on version 11, it seemed to break it and have the opposite effect. And even when I could tell that it was pinging the board (in version 0.10.2) it still went down unless I was there to constantly refresh the webpage. I can tell that it is installed correctly because of the info tab. ☹️ I don't know why it's doing this and I was wondering if you have any ideas?
Thanks!

RandomPotato893, 2020-12-03:
I wonder, Do I need to install any library's for this?

ulte, 2020-12-03:
I found that lowering the the ping delay (PingDelayMs) in the usermod_Fix_unreachable_netservices.h file from 60000 to 10000 worked for me. 60 secs was too long in my environment for some reason

huggy-d1, 2020-12-03:
> I wonder, Do I need to install any libraries for this?

If you use VS Code with PlatformIO IDE, it should pick up any required libraries.  That said, I do not think the "Fix" usermod requires additional libraries.

RandomPotato893, 2020-12-03:
@ulte Thanks! I'll try that @huggy-d1 I have that exact setup with vscode so if what @ulte said doesn't work, I'll be really confused. Thanks for your help everyone @ulte I'm trying that right now.

RandomPotato893, 2020-12-04:
@ulte I tried that and it didn't work I am very confused.

RandomPotato893, 2020-12-04:
Ok so now I'm at a new level of problem. Even pinging it doesn't help and I know that I am pinging it correctlly because I am doing it with command prompt. The webpage just dies whether or not I ping it. I have no idea what is going on.

gegu, 2020-12-10:
>
>  I know that I am pinging it correctlly because I am doing it with command prompt

Pinging the device from command prompt unfortunately does not help with the unreachable net-services problem. At least with my configuration I tried that too. Here an outgoing connection is required as a workaround.

I have updated the [usermod_Fix_unreachable_netservices.](https://github.com/Aircoookie/WLED/tree/master/usermods/Fix_unreachable_netservices_v2) The delay between pings can now be configured in the web UI and the setting is persistent after a reboot.

You can try setting the delay to 10 seconds, maybe it will help.

closed 2021-06-16 by stale[bot]