wled/WLED #1410: Increase number of Time-controlled presets

jamesmyatt, 2020-12-01:
Now we're using LittleFS for config, would it be possible to increase the number of time-controlled presets that you can define? Would it even be possible to make the length of the list variable now?

Aircoookie, 2020-12-01:
As far as the filesystem goes, making it variable is of course possible. Each entry is still stored in RAM though, so more timed presets will take up more. Maybe we can make that dynamic as well though!
Will get to this once I revamp the settings pages, perhaps that is also a good opportunity to support dates and more precise definition of recurrency (perhaps with your idea of cron syntax in #1411)

jamesmyatt, 2020-12-01:
Great. Thanks.

sheppp1, 2023-10-28:
Aircoookie: Add one more to the list of those that would like the capability to add more time-controlled presets - especially now that you have integrated the calendar functionality into it. There are far more days of the year with various color schemes than there are presets.
Furthermore, you should integrate the sunrise/sunset into each preset now that there is calendar functionality.

zowper, 2024-04-25:
I would also love to see both of these features added!

1- Rather than a chart, it would be great to have an option to simply add a row based on your needs (maybe with the available RAM displayed nearby).
2- I also love the idea of choosing between either an absolute time *or* a relative time based on sunrise/sunset for each preset (as mentioned by sheppp1).

Thank you all!  You are all amazing and I love the WLED community!!  😁🙌🎉

blazoncek, 2025-12-29:
#5140 will resolve this.