wled/WLED #1414: Default settings not retained when power cycling the power supply v.11.0 vs. v0.8.5

csantelman, 2020-12-01:
I'm confused - I still can't get this to work. In my case, I want to use a wifi enabled switch to control the power supply, which would then control the lights.

Using the wled app on my phone - I set the lights to be a non-default color - blue.
Navigate to config/LED Preferences, check the "Set Current preset cycle setting as boot default" box
Save (Settings saved message appears)
Power off the lights via the phone app, power back on, and the light is blue. This is expected behavior.
Turn off the wifi switch, lights go off (expected)
Turn on the wifi switch, lights come on, but are back to the defualt color (unexpected)

**WLED version**
 - Board: [ESP8266]
 - Version [0.11.0, 2011290]
 - Format [Binary]

**Additional context**
I flashed my device back to v0.8.5 revision, and the led configuration screen had the following selection "Set current color, brightness and effect as boot default" selection.  This works in my above scenario.

This options appears to be missing in the latest web ui.

Aircoookie, 2020-12-01:
Hi! To set a preset as boot default, save it in Favorites tab in the main UI, note its ID (you can find it in the bottom of the preset) and set that ID as `Boot preset` in LED settings. That will do what you want :) The  "Set Current preset cycle setting as boot default" is only for when you have a preset cycle active, which is multiple presets that automatically change.

csantelman, 2020-12-01:
@Aircoookie - that resolved my issue.  Thanks for your prompt reply, much appreciated.

closed 2021-06-03 by Aircoookie