wled/WLED #1415: Add sorting for the preset list

danecreekphotography, 2020-12-01:
**Is your feature request related to a problem? Please describe.**
The new preset feature is wonderful but the list appears to be sorted in random order. 

**Describe the solution you'd like**

It'd be great to offer two sort options:

1. Sort by preset name
2. Sort by preset slot number

Aircoookie, 2020-12-01:
Hi! What browser/setup do you use? The list should be sorted in alphabetical order...
Adding an option to sort the list by ID is a good idea though, I might add that.

danecreekphotography, 2020-12-01:
You know, you're right, they are and my naming convention is just stupid. I have some with dashes in the middle and some without that made it look like they weren't sorted properly. 🤦‍♂️

If you leave this open for a bit I might take a crack at adding the sort by ID option, shouldn't be too hard to do on the HTML side of things.

nlflint, 2020-12-01:
Because the Preset Cycle is based on a range of IDs, it's important to see presets in their ID order.

Came here to suggest something: Prefixing the names with their ID. I did this by hand in my presets.

Example:

Given Preset #24 is named "Dancing Lights"
Then the preset title is:`24. Dancing Lights`.

[image]

danecreekphotography, 2020-12-01:
I was thinking something similar as well! I might take a crack at it.

blazoncek, 2023-01-25:
The presets are sorted alphabetically and have search bar at the top in 0.14 or later.

closed 2023-01-25 by blazoncek