wled/WLED #1417: NODE MCU no longer connecting

bluedke70, 2020-12-02:
I had to replace several splices and strings in my house run this year, but once I did that I was bale to get them working.  I refreshed two NODE MCU 2866's with WLED10.2 and now I can not find the boards anymore.  I was able to initially access WLED-AP and I could control one of the boards, but not the other.  Now I cant discover either board.  When I plug the power supply in the lights all light up orange, like the default setting.

I added a wreath to the setup.  I had the wiring set up to run the roof line on one mcu and the wreaths on the other.  Now I cant discover the node mcu or wled.  I have reflashed the boards, twice and I read the voltage out of pin D4 on both and its 1.3VDC but its going through a 3.3V to 5V logic level shifter.

Any help would be appreciated.
Thank you for your help!

pk88888, 2020-12-03:
Hi! just try to flash with ESPHome-Flasher. I have the same problem now and if i flash with NodeMCU WLED-AP doesnt appear. If i flash with ESPhome it works normal. I dont know why... ;)

bluedke70, 2020-12-04:
Thanks fir the help. I did that and I got the roof line working but I cannot get the second board to work at all. I’ve looked at my router and it’s not showing up on the wireless network either. So I assume this is a board issue. ?  I’ve tried two different boards   Is there a setting in missing?  WLED used to discover the two boards now it doesn’t.

bluedke70, 2020-12-15:
It was the way my router recognized the board. I fixed it. Thanks for everyone’s help and input

closed 2020-12-15 by bluedke70

pk88888, 2020-12-15:
How did you fix it?

bluedke70, 2020-12-15:
I triple checked my wiring and my voltages and grounds and everything was fine so I switched places for the note MC use and the wreath that we’re not working on the one note MCU when I’m working and the other one was working in default mode but I could not control it… So I went back to my router and entered a number and was able to miraculously get one of the numbers that was not assigned to work… Later on that number fell off and I was able to put it on a different number that was working from before and it solid so far… Now the only hat problem I have is with some flashing if I try to go up on the brightness… From the peak over it starts to flash and the opposite side will flash down as it flashes up but if I keep the brightness down it works fine

pk88888, 2020-12-15:
It's interesting, I see the same problem with resetting the device when the brightness is set to the maximum position. I will have to try to fix this problem soon.