wled/WLED #1418: All effects an solid colours not talking to LEDs correctly

pbezza, 2020-12-02:
I have installed the v11.0 bin on to my QuinLED-Dig-Quad with an Esp 32 but it not talking to the LEDs correctly. Choosing peek in the GUI is also wrong and matches what my LEDs are showing. I have 6 5m lengths of ws2812b 50 per length. The first 54 LEDs barely show any colour and the last 50 show nothing. If I add more LEDs, they also show nothing. As the peek is identical to the physical LEDs, I am positive that my LEDs are not at fault and have swapped out these LEDs and yielded the same results.
[image] 
[image]

pbezza, 2020-12-03:
Tried Quindor's v0.11.0 bin files but have the same problem. Switching to an older version to see if the problem goes away.

Aircoookie, 2020-12-03:
Hi, sorry for the issue! Others have had this problem as well, it can happen if some flash contents are invalid.
It often helps to erase the flash completely (for example with `esptool.py erase_flash`, your flasher might also have such an option) and then reinstall the binary.

pbezza, 2020-12-03:
Thanks for replying. I'll try tonight after work. It was a brand new esp32 which should have been blank already.

sansillusion, 2020-12-04:
Inhave had weird issues mainly stability, I flashed multiple versions and multiple options activated/deactivated etc... Used usb and ota, when using USB I often chose erase data option (having to reconfig wifi, leds etc...).

Lately I wanted to flash a bin (before 0.11) and looked at the quickstart guide. I decided to flash the provided bootloader, when it rebooted I noticed it had reset everything. So I reconfigured everything before trying to flash the bin. Weirdly with the save wled I started havin 190kb free mem instead of 170kb I decided to keep it like that and it's been working great ever since. I compiled latest source a few days ago and it worked like a charm without any bugs anymore ! Could the bootloader from Aircoookie be somehow better than what my Arduino IDE was giving me (esp32 intalled trough git master aswell as wled, except since 0.11.0 wich I had to install and use vscode/pio)

pbezza, 2020-12-04:
I also used the bootloader from the quickstart guide. Unfortunately my controller is inside my roof cavity so I have to crawl up there again to grab the esp32 so I can wipe and flash again. Fingers crossed.

pbezza, 2020-12-04:
I flashed v0.10.2 over v0.11.0 and it worked flawlessly. Flashed v0.11.0 again and had the same issue.

I then selected and applied 'Factory Reset' under 'Security and updates' and v0.11.0 now works perfectly. All fixed over WiFi.

closed 2020-12-04 by pbezza