wled/WLED #1423: Analg Clock overlay settings no longer saved across reboots

bgutman, 2020-12-02:
**Describe the bug**
After upgrading to 0.11.0 setting Clock overlay First LED, Last LED, 12h LED, Show 5min marks, Seconds (as trail) do not save across reboots

**To Reproduce**
In Time Settings, select Analog clock overlay, change any/all values from default, save, then reboot or power cycle

**Expected behavior**
Settings to be saved across reboot

**WLED version**
 - Board: Wemos D1 
 - Version WLED 0.11.0 (build 2011290)
 - Format Binary

**Additional context**

Thank you for your help!

bgutman, 2020-12-04:
Also confirmed same behavior on a ESP-Wroom-32

gegu, 2020-12-11:
I have also reported this problem a little later. It was fixed with #1448

closed 2020-12-12 by bgutman