wled/WLED #1425: demo effect

giannimp, 2020-12-02:
All effect in automatic??

**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2020-12-03:
Sorry, I don't really understand. If you want a demo mode that cycles through all effects, that doesn't really make sense since there are far too many effects in WLED...

sansillusion, 2020-12-03:
I think it would make sense. I sometimes would like it to cycle through
effects (and palettes). We could use the intensity slider for random or
cycle (1-128 = cycle, 129-256 = random) and speed slider for delay between
changes.

Legsmaniac, 2020-12-03:
Surely there must be a way this could be added as a custom UserMod for those who want to try it? 🙂

zimmer62, 2020-12-03:
> Surely there must be a way this could be added as a custom UserMod for those who want to try it? 🙂

I think a usermod with it's own page would be great for this.  A speed selector and a couple other options...

Go through each effect, while displaying which one on the screen so you could see them all and remember which is which.  I bet I could whip this up this coming weekend if I can find enough alone time.

huggy-d1, 2020-12-03:
I agree it would be a fantastic usermod.

To see the current effect / palette on a WLED web page, and also a way to adjust AND automatically save settings.  As others have said, colors / speed / intensity, and automatically save those settings as the usermod switches to next effect / palette.

When it runs around and starts over, it restores existing settings as it runs through effects / palettes again.

This way, we could fine tune each effect / palette.

Perhaps display JSON for those settings as they are restored / adjusted.

At some point it starts to get too complicated, but having every effect / palette configured exactly the way you'd like them, and not being able to capture that would be such a shame (waste).

zimmer62, 2020-12-04:
I made a program to copy my presets from one wled to another, this is how I figured it would work.

I've written a usermod tonight that loops through all of the effects.  

I don't think it will be very hard to write a web page that shows what effect is currently displaying, give you some options to adjust the speed etc...  

BUT....  I don't know how much time I'll actually have to make it really nice.

setup the fx variable in the private section:
int fx = 0;

then do this in your loop and boom you've got every effect for 10 seconds.

void loop() {
      if (millis() - lastTime > 10000) {
        Serial.println("I'm alive!");
        lastTime = millis();
        strip.setMode(strip.getMainSegmentId(), fx);
        fx++;
        if (fx > MODE_COUNT) fx = 0;
      }
    }

closed 2021-06-16 by stale[bot]