wled/WLED #1426: With the power button, NoteMCU goes off but you cannot turn it on again

Hardwarekiller, 2020-12-03:
Since version 11.0 the power button cannot be used to switch the LED strip on again after it has been switched off.
What is also strange is that when you press the power button on the V11.0, the blue service LED on the NoteMCU goes off. That was not the case with the 10.2.
Switching on the LED strip is no longer possible after switching it off in the host with the power button, since the LED strip can no longer be switched on with the power button.
Settings in Wled go however also the "peek" display works.
But the LED strip remains dark and the blue service LED is also off
However, you can go into the config in wled and then under "updates" then click on reset then wled reacts again

Aircoookie, 2020-12-03:
Hi!
The blue LED turning off is intended :) It was annoying for many users, including me.
The power button issue you are reporting is very strange. I assume you mean the power button in the top left of the WLED UI (or do you mean the hardware button you can connect to gpio0? Or the one in the app device overview?)
All of them work fine for me, so I can't reproduce it... Have you tried using the button multiple times?

If you'd like, you can send me your `[WLED-IP]/cfg.json` and `[WLED-IP]/presets.json` files and I'll have a look if it is something configuration specific (don't worry, cfg.json does not include your wifi password)

Hardwarekiller, 2020-12-04:
""I assume you mean the power button in the top left of the WLED UI"
Yes, i mean this !

""If you'd like, you can send me your [WLED-IP]/cfg.json and [WLED-IP]/presets.json files and I'll have a look if it is something configuration specific (don't worry, cfg.json does not include your wifi password)""
How can I create the 2 files?
I didn't wled compiler, I just flashed the bin file

Should I create a video on YouTube and upload it?

Thanks 
Hardwarekiller

sansillusion, 2020-12-04:
What he means is to point your browser to these adresses replacing
[WLED-IP] by the IP of your device... ie : 192.168.1.101/presets.json

Hardwarekiller, 2020-12-05:
Ok.
cfg.json
{"rev":[1,0],"vid":2011290,"id":{"mdns":"wled-Node1","name":"WLED","inv":"Light"},"nw":{"ins":[{"ssid":"Killer1","pskl":32,"ip":[0,0,0,0],"gw":[0,0,0,0],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP","pskl":8,"chan":1,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":true,"phy":1},"hw":{"led":{"total":32,"maxpwr":1500,"ledma":55,"rev":false,"ins":[{"en":true,"start":0,"len":32,"pin":[2],"order":0,"rev":false,"skip":0,"type":22}]},"btn":{"ins":[{"type":2,"pin":[0],"macros":[1,2,3]}]},"relay":{"pin":12,"rev":false},"status":{"pin":-1}},"light":{"scale-bri":100,"pal-mode":0,"gc":{"bri":2.8,"col":2.8},"tr":{"mode":true,"dur":7,"pal":0},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":1,"on":true,"bri":128},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"twice":false}},"live":{"en":true,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"addr":1,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":true,"macros":[0,0]},"blynk":{"token":""},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-30c09c","topics":{"device":"wled/30c09c","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[192,168,178,0]},"ntp":{"en":true,"host":"0.wled.pool.ntp.org","tz":2,"offset":0,"ampm":false}},"ol":{"clock":0,"cntdwn":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{}}

presets.json
{"0":{},"1":{"n":"Preset 1","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":74,"sx":225,"ix":233,"pal":0}},"2":{"n":"Preset 2","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":90,"sx":225,"ix":233,"pal":0}},"3":{"n":"Preset 3","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":66,"sx":225,"ix":233,"pal":0}},"4":{"n":"Preset 4","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":87,"sx":225,"ix":36,"pal":0}},"5":{"n":"Preset 5","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":4,"sx":61,"ix":255,"pal":0}},"6":{"n":"Preset 6","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":74,"sx":5,"ix":9,"pal":0}},"7":{"n":"Preset 7","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":89,"sx":145,"ix":98,"pal":0}},"8":{"n":"Preset 8","bri":255,"seg":{"col":[[255,17,0],[0,34,255],[0,0,0]],"fx":10,"sx":248,"ix":51,"pal":0}},"9":{"n":"Preset 9","bri":255,"seg":{"col":[[255,255,255],[0,0,0],[0,0,0]],"fx":20,"sx":255,"ix":196,"pal":0}},"10":{"n":"Preset 10","bri":255,"seg":{"col":[[255,255,255],[0,0,0],[0,0,0]],"fx":23,"sx":253,"ix":255,"pal":0}},"11":{"n":"Preset 11","bri":255,"seg":{"col":[[255,0,0],[0,0,255],[0,0,0]],"fx":36,"sx":255,"ix":255,"pal":0}},"12":{"n":"Preset 12","bri":255,"seg":{"col":[[255,0,0],[0,0,255],[8,255,0]],"fx":54,"sx":255,"ix":255,"pal":0}},"13":{"n":"Preset 13","bri":255,"seg":{"col":[[255,17,0],[0,0,0],[0,0,0]],"fx":74,"sx":5,"ix":9,"pal":0}},"14":{"n":"Preset 14","bri":255,"seg":{"col":[[255,0,0],[0,0,255],[8,255,0]],"fx":48,"sx":203,"ix":193,"pal":0}},"15":{"n":"Preset 15","bri":255,"seg":{"col":[[255,0,0],[0,0,255],[8,255,0]],"fx":65,"sx":58,"ix":246,"pal":0}},"16":{"n":"Preset 16","bri":255,"mainseg":0,"seg":[{"id":0,"start":0,"stop":32,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,0,0],[8,255,0],[0,0,255]],"fx":69,"sx":234,"ix":232,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]},"17":{"n":"Z Macro 1","win":"FX=~"},"18":{"n":"Z Macro 2","win":"FX=~-"},"19":{"n":"Z Macro 3","win":"PL=~"}                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              ,"21":{"n":"Preset 18","on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":32,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,17,0],[0,0,0],[0,0,255]],"fx":104,"sx":222,"ix":240,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]},"20":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":32,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,17,0],[0,0,0],[0,0,255]],"fx":60,"sx":222,"ix":240,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Preset 17"}}

Thanks

Hardwarekiller

Avdbergnmf, 2020-12-14:
I have the exact same issue! Which is weird because with a different strip it doesn't happen, with the exact same setup (I believe). Have you figured out what the issue is? 

To add onto the bug, the same thing happens when you dim the LEDs to (near) 0 brightness. They will not turn back on...

sansillusion, 2020-12-14:
What power source are you using ?
I had a power pack that would turn the output off if under 500ma.

Aircoookie, 2020-12-14:
I think I have managed to reproduce this (independent from this issue) and already fixed it in latest master!
For me it occured only if the "Gamma Correct Brightness" option is enabled (I didn't catch the bug earlier since I usually don't use that option)

@Hardwarekiller from your log from a few days ago I can indeed see that you have brightness gamma correction enabled. Turning it off or installing master/waiting for 0.11.1 will fix it :)

Avdbergnmf, 2020-12-15:
@Aircoookie , you' re a legend. I managed to fix it by resetting the WLED installation. But that also disables that option, so it was probably the same issue!

Hardwarekiller, 2020-12-17:
@sansillusion

No that's not the reason.
My power supply has a very stable 5v and can deliver far more current than is necessary.
I come from electronics development and also write electronics software in Github.
The problem is clearly a software bug.

If you create profiles and delete profiles and add marros, the __presets.json__ file will be destroyed.
Have checked that several times
What I noticed is that this also happens in part in 10.5, but not in the preset file
But the timer function that is under the preset file doesn't work anymore

@Aircoookie 

There seems to be a lot more to be a bug. I tested something on the weekend
If you create profiles and delete profiles and add marros, the __presets.json__ file is destroyed.
Have checked that several times
What I noticed is the timer function that can no longer be changed under the preset file
Which is then noticed in the __presets.json__ file
that the 3 square brackets at the end of the file are no longer connected but the 3th square bracket is individually over several 100 empty spaces further away

Hardwarekiller, 2021-06-04:
works

closed 2021-06-04 by Hardwarekiller