wled/WLED #1429: New Presets and JSON API

apop880, 2020-12-03:
Apologies if I missed this, but are there plans to add the new preset functionalities introduced in 0.11 to the JSON API? Specifically, saving a preset with a name, seeing the name of the currently loaded preset, and potentially seeing a list of all presets alongside the lists of effects and palettes?

Aircoookie, 2020-12-04:
Hi, thank you for the heads up! This is actually part of the API already (UI uses JSON api to communicate with board), but I totally forgot to document it!

In short:
- Save a named preset with `{"psave":<preset number here>;"n":"Preset name","ib":true,"sb":true}`
- Obtaining the preset list is possible with the `/presets.json` file 
- Seeing the name of the current preset with the API only is not possible, you can only obtain its ID. Then you'd need some code to match it with the entries from presets.json (this is also what the UI does)

huggy-d1, 2020-12-04:
@Aircoookie added your "in short" comments to wiki in presets section.

apop880, 2020-12-09:
@Aircoookie Thank you! I figured it had to be in there somewhere so the UI could use it. One last question: when applying a preset via the API, I assume I'd also need to first take the name to presets.json to fetch the ID, and then populate the API call with the ID?

Aircoookie, 2020-12-09:
Yes, you'd need to have the presets.json to find the ID for the name. Via the API, you can only apply presets by ID with the `{"ps":<id here>}` command :)

closed 2020-12-16 by apop880