wled/WLED #1434: 'type error cannot set property of inner HTML of null'

luminxa, 2020-12-04:
'type error cannot set property of inner HTML of null' I'm getting this error on the homepage load.. Everything else seems to be working... any ideas?

Aircoookie, 2020-12-04:
Hi! Do you know what you did when the error occured?
You can try opening the developer tools with `Ctrl + shift + i` and take a look at the console in the bottom right. It should have some info about the error that helps me understand what the problem is :)

luminxa, 2020-12-04:
I'm sorry I should have said this is only happening on mobile so I can't really open up the developer tools.. I'm wondering if my browsers are out of date on my mobile devices and it's kicking off some sort of errors

luminxa, 2020-12-04:
Also wanted to ask you with the Cs data. JS file that minimizes automatically the UI is it stripping anything from the code?

luminxa, 2020-12-04:
I think it might be the customizations I was trying to add it was a CSS animated background.. I'm going to try to revert and see if it works without the background

luminxa, 2020-12-04:
Well I don't think it was the background because I reverted and I'm still getting the error and I found this in the developer console

Uncaught (in promise) TypeError: Cannot set property 'innerHTML' of null
    at presetError (sliders:354)
    at sliders:382

luminxa, 2020-12-04:
From what I can tell the air appears to be happening because it can't load the presets Json file?

luminxa, 2020-12-04:
It appears it was something in the in my code I think I have it fixed

closed 2020-12-05 by luminxa