wled/WLED #1437: [Preset] can't remove "include brightness" option on old or new preset

cadavor, 2020-12-05:
Hello everyone,
I've updated my ESP32 WLED 0.8.3 to latest version (0.11) , all my preset are still present (cool).

I've tried to use the new feature "include brightness" because I want to remove it on all my preset.
When I edit old presets, it only shows the API command textbox but I didn't find what to write in for removing brightness (just removing the bri value?)
`{"bri":134,"seg":{"col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":142,"ix":128,"pal":0}}`

So I tried to overwrite one preset by : 
- load the preset by selecting it
- create a new preset with "use current state" checked and uncheck "include brightness"
- save it on the same id than selected preset to overwrite it
but the "include brightness" option is ON on the saved preset...

So I tried to create a new preset without overwritting existent one (new ID) but same problem, "include brightness" option still stuck ON.

I make the test on my smartphone and on my PC with Chrome and Firefox.

Someone experience the same problem? Is this a bug?

Aircoookie, 2020-12-07:
Hi and thank you for the report!
This likely is not a bug (if it is I can't reproduce), but unintuitive usability (sorry about that!)
Firstly, you are correct that if you want to remove the brightness from an existing segment, you can just remove the `"bri"` key from the JSON command and save.
You can also re-save the preset like you described and it _should_ have saved the preset without the brightness included. By refreshing the page and looking if the API command contains the bri key you can see if it worked.

However, here the usability issue comes in. The "Include brightness" checkmark is always on by default (it will only have an effect though if you re-save the preset). Maybe I can have this disable (same for the "Save segment bounds") when the brightness is not included in the API command. Not an issue from a technical standpoint, but I can understand that you would have expected that checkmark to stay unchecked.

cadavor, 2020-12-07:
Thank you for your detailed answer.
Indeed I was expecting the checkbox to remain unchecked.
I remove the "bri" key in JSON command box and save and it work.

You can close this issue if you want

closed 2021-06-16 by stale[bot]