wled/WLED #1438: Home assistant taking down webpage

RandomPotato893, 2020-12-05:
**Describe the bug**
I’ve been troubleshooting this software for about a week. My router (which is an amplifi instant wifi) was taking my sever down because of an issue with static IP addresses. I fixed that issue and the server started staying up. Then I tried to add it to home assistant with the home assistant plugin and the webpage went offline.  I have a similar post on https://wled.discourse.group Any ideas?

**WLED version**
 - Board: [D1 mini]
 - Version [0.11.0]
 - Format [Binary self compiled]

Aircoookie, 2020-12-05:
Forum post for reference: https://wled.discourse.group/t/home-assistant-integration-breaking-my-wled/1559/3

I don't really have a clue unfortunately :( Wifi issues are always the hardest to track down once the worst offenders (signal strength, 2.4Ghz) turned out not to be the culprit.

RandomPotato893, 2020-12-05:
I don't know what to do then can you control effects via mqtt?

closed 2021-06-16 by stale[bot]