wled/WLED #1441: Some effects crashing and error in log after flashing "corrupted dir pair at {0x0, 0x1}"

Justdigit, 2020-12-05:
On latest WLED flased on a ESP32 WLED 0.11.0 (build 2011290)

Some effects crash while selecting. (failed to fetch) It looks likes to me when you change effects fast after each other.

Also, when i have flashed with my favorite downloader i get an error message from LittleFS as included. 
I don't know if it has something to do  with it.

`Showing logs:
[22:36:18].pio\libdeps\esp32dev\LittleFS_esp32\src\lfs.c:1003:error: Corrupted dir pair at {0x0, 0x1}
[22:36:18]E (59) esp_littlefs: mount failed,  (-84)
[22:36:18]E (60) esp_littlefs: Failed to initialize LittleFS
[22:36:19]Ada
[22:44:07]dhcps: send_offer>>udp_sendto result 0
[22:44:52]dhcps: send_offer>>udp_sendto result 0
[22:45:48]ets Jun  8 2016 00:22:57
[22:45:48]
[22:45:48]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:45:48]configsip: 0, SPIWP:0xee
[22:45:48]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:45:48]mode:DOUT, clock div:2
[22:45:48]load:0x3fff0018,len:4
[22:45:48]load:0x3fff001c,len:1044
[22:45:48]load:0x40078000,len:8896
[22:45:48]load:0x40080400,len:5828
[22:45:48]entry 0x400806ac
[22:45:49]Ada
[22:46:07]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:46:07]Core 1 register dump:
[22:46:07]PC      : 0x400dcbdb  PS      : 0x00060630  A0      : 0x800dcc2b  A1      : 0x3ffb1e10  
[22:46:07]A2      : 0x00000000  A3      : 0x00000000  A4      : 0x8008a560  A5      : 0x3ffd49e0  
[22:46:07]A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0x00000020  A9      : 0x80000020  
[22:46:07]A10     : 0x00000000  A11     : 0x3ffd4d00  A12     : 0x00000005  A13     : 0x00000020  
[22:46:07]A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
[22:46:07]EXCVADDR: 0x00000000  LBEG    : 0x4018b162  LEND    : 0x4018b177  LCOUNT  : 0x00000000  
[22:46:07]
[22:46:07]Backtrace: 0x400dcbdb:0x3ffb1e10 0x400dcc28:0x3ffb1e40 0x400dd1ee:0x3ffb1e70 0x400dd2fe:0x3ffb1ea0 0x400de2fa:0x3ffb1ec0 0x400d61fd:0x3ffb1f10 0x400ded28:0x3ffb1f40 0x400f5d56:0x3ffb1f60 0x400f654a:0x3ffb1f90 0x40112129:0x3ffb1fb0 0x4008a34d:0x3ffb1fd0
[22:46:07]
[22:46:07]Rebooting...
[22:46:07]ets Jun  8 2016 00:22:57
[22:46:07]
[22:46:07]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:46:07]configsip: 0, SPIWP:0xee
[22:46:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:46:07]mode:DOUT, clock div:2
[22:46:07]load:0x3fff0018,len:4
[22:46:07]load:0x3fff001c,len:1044
[22:46:07]load:0x40078000,len:8896
[22:46:07]load:0x40080400,len:5828
[22:46:07]entry 0x400806ac
[22:46:07]Ada
[22:46:42]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:46:42]Core 1 register dump:
[22:46:42]PC      : 0x400dcbdb  PS      : 0x00060030  A0      : 0x800dcc2b  A1      : 0x3ffb1e40  
[22:46:42]A2      : 0x00000000  A3      : 0x00000000  A4      : 0x8008a560  A5      : 0x3ffd49d0  
[22:46:42]A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0x00000020  A9      : 0x80000020  
[22:46:42]A10     : 0x00000000  A11     : 0x3ffd4cf0  A12     : 0x00000005  A13     : 0x00000020  
[22:46:42]A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[22:46:42]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[22:46:42]
[22:46:42]Backtrace: 0x400dcbdb:0x3ffb1e40 0x400dcc28:0x3ffb1e70 0x400dd1ee:0x3ffb1ea0 0x400dd2fe:0x3ffb1ed0 0x400d7a80:0x3ffb1ef0 0x400ded28:0x3ffb1f40 0x400f5d56:0x3ffb1f60 0x400f654a:0x3ffb1f90 0x40112129:0x3ffb1fb0 0x4008a34d:0x3ffb1fd0
[22:46:42]
[22:46:42]Rebooting...
[22:46:42]ets Jun  8 2016 00:22:57
[22:46:42]
[22:46:42]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:46:42]configsip: 0, SPIWP:0xee
[22:46:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:46:42]mode:DOUT, clock div:2
[22:46:42]load:0x3fff0018,len:4
[22:46:42]load:0x3fff001c,len:1044
[... 63 lines cut ...]
[22:59:44]Rebooting...
[22:59:44]ets Jun  8 2016 00:22:57
[22:59:44]
[22:59:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:59:44]configsip: 0, SPIWP:0xee
[22:59:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:59:44]mode:DOUT, clock div:2
[22:59:44]load:0x3fff0018,len:4
[22:59:44]load:0x3fff001c,len:1044
[22:59:44]load:0x40078000,len:8896
[22:59:44]load:0x40080400,len:5828
[22:59:44]entry 0x400806ac
[22:59:45]Ada
[22:59:53]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:59:53]Core 1 register dump:
[22:59:53]PC      : 0x400dcbdb  PS      : 0x00060e30  A0      : 0x800dcc2b  A1      : 0x3ffb1e00  
[22:59:53]A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbf178  A5      : 0x00000001  
[22:59:53]A6      : 0x0000002a  A7      : 0x3ffd4880  A8      : 0x00000000  A9      : 0x3ffb1e3d  
[22:59:53]A10     : 0x00000000  A11     : 0x00002280  A12     : 0x00000001  A13     : 0x3ffd7cc6  
[22:59:53]A14     : 0x0000000a  A15     : 0x3ffd4850  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
[22:59:53]EXCVADDR: 0x00000000  LBEG    : 0x4018b162  LEND    : 0x4018b177  LCOUNT  : 0x00000000  
[22:59:53]
[22:59:53]Backtrace: 0x400dcbdb:0x3ffb1e00 0x400dcc28:0x3ffb1e30 0x400dd1ee:0x3ffb1e60 0x400dd2fe:0x3ffb1e90 0x400de2fa:0x3ffb1eb0 0x400d7689:0x3ffb1f00 0x400ded28:0x3ffb1f40 0x400f5d56:0x3ffb1f60 0x400f654a:0x3ffb1f90 0x40112129:0x3ffb1fb0 0x4008a34d:0x3ffb1fd0
[22:59:53]
[22:59:53]Rebooting...`

Aircoookie, 2020-12-07:
Hi and thank you for the log! Very likely same issue as #1431 
What you can try is format the littlefs by going into Security settings and doing a factory reset.
According to this: https://github.com/littlefs-project/littlefs/issues/476#issuecomment-727245410 the corrupted dir pair error might mean that the filesystem could not find/access its memory area. What model of ESP32 do you have?

yetdog, 2020-12-07:
This is great info to see - awesome that layers are being peeled back on this. One of the downsides to these little boards - no good real "operating environment" to jump into and dump logs from (easily, at least). Thanks @Justdigit and @Aircoookie 

For the record ive got the "MELIFE ESP32 ESP-32S Dev Board" from amazon: https://www.amazon.com/gp/product/B07Q576VWZ/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

Justdigit, 2020-12-07:
> Hi and thank you for the log! Very likely same issue as #1431
> What you can try is format the littlefs by going into Security settings and doing a factory reset.
> According to this: [littlefs-project/littlefs#476 (comment)](https://github.com/littlefs-project/littlefs/issues/476#issuecomment-727245410) the corrupted dir pair error might mean that the filesystem could not find/access its memory area. What model of ESP32 do you have?

Its an ESP32 wroom board from AE. So yeah maybe its a corrupted memory part.

Justdigit, 2020-12-08:
@Aircoookie Received a new ESP32. But it has the same problem. Tried it on an esp8266 and there is no error in the Log.
But i need the ESP32 because of the multipin mod. And i cant get it to work on esp8266

Device info.

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES

tsctrl, 2021-02-19:
Hi Is this resolved? 

i have the same issue. just using the base LITTLEFS_test example sketch to reproduce with esp32 devkitC v1

LittleFS_esp32\src\lfs.c:1003:error: Corrupted dir pair at {0x0, 0x1}

Justdigit, 2021-02-20:
> 
> 
> Hi Is this resolved?
> 
> i have the same issue. just using the base LITTLEFS_test example sketch to reproduce with esp32 devkitC v1
> 
> LittleFS_esp32\src\lfs.c:1003:error: Corrupted dir pair at {0x0, 0x1}

Yeah.... i guess you flash by ESP Home  flasher? its nog giving the error code by Visual Studio when compling yourself.

closed 2021-02-22 by Justdigit