wled/WLED #1442: Cannot build with usermod PIR sensor switch

marulitua, 2020-12-06:
**Describe the bug**
I want to add PIR to my setup, so I follow [this documentation](https://github.com/Aircoookie/WLED/tree/v0.11.0/usermods/PIR_sensor_switch), but cannot pass the linking process

**To Reproduce**
`cp usermods/PIR_sensor_switch/usermod_PIR_sensor_switch.h wled00/.` and register the usermod

**Expected behavior**
Should be able to build

**WLED version**
 - Board: esp32 devkit v1
 - Version v0.11.0
 - Format self-compiled

**Additional context**
This is the error message
```
Processing esp32dev (board: esp32dev; platform: espressif32@2.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <AsyncTCP> 1.0.3
|-- <IRremoteESP8266> 2.7.3
|-- <LittleFS_esp32> 1.0.5+sha.b941415
|   |-- <FS> 1.0
|-- <ESP Async WebServer> 2.0.0+sha.3336275
|   |-- <AsyncTCP> 1.0.3
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Building in release mode
*** use existing my_config.h ***
Compiling .pio/build/esp32dev/src/wled00.ino.cpp.o
Linking .pio/build/esp32dev/firmware.elf
.pio/build/esp32dev/src/usermods_list.cpp.o: In function `PIRsensorSwitch::newPIRsensorState(bool, bool)':
/media/eins/repos/WLED/wled00/fcn_declare.h:198:(.iram1[PIRsensorSwitch::ISR_PIRstateChange()]+0x3): dangerous relocation: l32r: literal placed after use: .literal._ZN15PIRsensorSwitch18ISR_PIRstateChangeEv
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
================================================================= [FAILED] Took 9.76 seconds =================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:09.758
=========================================================== 1 failed, 0 succeeded in 00:00:09.758 ===========================================================
```

This is my `wled00/usermods_list.cpp`
```
#include "wled.h"
/*
 * Register your v2 usermods here!
 *   (for v1 usermods using just usermod.cpp, you can ignore this file)
 */

/*
 * Add/uncomment your usermod filename here (and once more below)
 * || || ||
 * \/ \/ \/
 */
//#include "usermod_v2_example.h"
//#include "usermod_temperature.h"
//#include "usermod_v2_empty.h"
#include  "usermod_PIR_sensor_switch.h"

void registerUsermods()
{
  /*
   * Add your usermod class name here
   * || || ||
   * \/ \/ \/
   */
  //usermods.add(new MyExampleUsermod());
  //usermods.add(new UsermodTemperature());
  //usermods.add(new UsermodRenameMe());
  usermods.add(new PIRsensorSwitch());

}
```

If I comment out [this line](https://github.com/Aircoookie/WLED/blob/3a3948e74fb4bd72e70f5516bd7296d98a3ab9e6/usermods/PIR_sensor_switch/usermod_PIR_sensor_switch.h#L57), I don't have linking problem.

What am I missing here ?

Thank you for your help!

Aircoookie, 2020-12-06:
Hi! Looks like the error happens in the interrupt routine. I am not the author of this mod, but I suspect it was only written for ESP8266 and thus does not build on the ESP32.

@gegu do you know what this could be?

gegu, 2020-12-06:
Yes, this is due to the interrupt. Unfortunately it does not compile with ESP32 and there is no easy solution. I don't have the time at the moment to dive deeper into the topic of ESP32 interrupts. Although I don't like it very much, I'll switch the handling to pin polling soon.

gegu, 2020-12-07:
@Aircoookie The problem was not the interrupts. The compiler complains about static method bodies in the class declaration. I have implemented them now outside.

@marulitua Please test if it works once it is merged (or build from my repo). I currently have no ESP32 installation available.

marulitua, 2020-12-08:
Hi @gegu ! Thanks for your update. I'll try it tonight. What do you mean by pin polling ? What does that mean ? Or can you point me to a documentation of something

gegu, 2020-12-08:
Pin polling means that the state of the pin is continuously checked for changes,. During interrupt handling a function is only called if the state changes.

marulitua, 2020-12-08:
Thank you for your explanation. Much Appreciated.
Your patch working as expected. I think we can close this issue.

closed 2020-12-08 by marulitua