wled/WLED #1443: Sync problems with 0.11 in both devices

panagiotis81, 2020-12-06:
Hi I have two WLED lamps which are synced. After I update both of them to 0.11 the sync is broken. Sometimes is works but most of the time my "slave" lamp does nothing.

With both lamps to 0.10.2 sync is fine
I kept my "master" lamp to 0.11 and I downgraded my "slave" lamp to 0.10.2 and everything works fine again.

I updated "slave" again to 0.11 and the problem was the same. And I downgraded again the slave one with 0.10.2 the sync works fine again.

**WLED version**
 - Board:  Wemos D1
 - Version: 0.11 / 0.10.2
 - Format: Binary

I have them also connected to Home Assistant through the integration to turn on/off and change effects/presets. When both run 0.11 there is also more delay.
The problem occurs both from HA and WLED web interface.

Thank you for your help!

Legsmaniac, 2020-12-06:
How very strange.
I'm not sure it's a bug because I'm using D1 with V11 installed and with WS2812B.  Sync works fine but I'm using Home Assistant with scripts for control instead which also works fine. Web interface is also ok for me.

Sorry this isn't helping you but just to say I can't reproduce on my setup.

panagiotis81, 2020-12-06:
It is weird.

I just tested a 3d d1 mini with a WS2812B ring and 0.11 (this won't normally sync with the lamps) and does the same.
I used the ring with 0.11 as master and see the same behavior. 
With more testing, I noticed that when all the devices are already on (I have turned them on separately)  the sync works (I change effects etc in all devices).
But if I turn them off, then turning on the "master" will only open the 0.10.2 device but not the other with 0.11.

I also use mainly scripts in HA to control with a drop down menu a couple of presets. But my scripts only call the "master". This is how I noticed that the sync is not working right after I updated them to 0.11

Aircoookie, 2020-12-07:
Hi!

> With more testing, I noticed that when all the devices are already on (I have turned them on separately) the sync works (I change effects etc in all devices).

Can you confirm that "Receive brightness" in the WLED broadcast section of Sync settings is enabled?

panagiotis81, 2020-12-07:
@Aircoookie
I don't think it's the settings.
On/Off button enabled: Yes
Receive Brightness,  Color, and Effects: All Yes

Also enabled "Send notifications twice" but doesn't change anything.

I thing the problem is 0.11 devices "waking up" from sync.

If I reboot the 0.11 "slave" device everything works fine for a while.

But after about 30min when I tested again, still the "slave" with 0.11 doesn't turn on (as the "slave" with 0.10.2 does) when master with 0.11 turns on. If I turn them on independently they can sync with "master" and can turn off when "master" turns off.

If 0.10.2 is "master" then the same problem occurs. The 0.11 devices don't turn on when "master" turns on.

sansillusion, 2020-12-07:
Have you tried disabling wifi sleep in the bottom of wifi config page ?

panagiotis81, 2020-12-07:
@sansillusion, @Aircoookie

You were right, disabling wi-fi sleep for 0.11 solves the problem. 
I never had this setting changed before in previous editions and everything worked fine. Something changed in this?

StefanRu1, 2020-12-08:
Hmm, that's really wired. 
I have two lamps in sync mode with 0.11. and they work just fine and i do not need to disable wifi sleep.
I can also control them over MQTT from FHEM and the sync works fine.
Power On / Off works also without a problem.
I am using Wemos D1 mini and both lamps are connected to the same access point.

BR,
Stefan

sansillusion, 2020-12-09:
The D1 mini is a esp8266, the esp32 variant seem to suffer wifi sleep
issues with some routers.

closed 2021-06-16 by stale[bot]