wled/WLED #1444: preset setting problem

mascolo74, 2020-12-06:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

mascolo74, 2020-12-06:
Board:  Wemos D1, ESP32 wrom]
version 0.11 installed
the problem I set the preset time to 3.5 s the preset starts to work, but if I reset, turn off, or restart, the cycle stops I saw that the time is set to 0
the check set to reboot the preset disappears

mascolo74, 2020-12-06:
I attach photos
[image]
[image]
[image]
[image]

Crosstuga, 2020-12-06:
Hi, I experience the same issue does not hold after reboot and but also something wrong in the segments in the preset cycle. Might be related on boot does not read the settings properly.

I have 2 segments defines and for both all settings for what I can see will get saved in the json file (http://<IP>/edit). If I click the preset "button" will do as set in both segments but if after reboot will treat it both segments as only one with total LED's - like segment 2 will not exist. The Segment 1 settings will play in 2 as well. If I force and click the button of the preset (or via API PL=~ for example) will play fine- It is like on reboot does not recognize the 2 segments. Sometimes need to refresh the browser to appear - Firefox last version on windows 10.

Thanks for any help. Outstanding piece of software btw!

Aircoookie, 2020-12-07:
@mascolo74 thank you for the report! It is intended that the cycle turns off when toggling the power or rebooting, however it should not set the time per preset to `0`. I will need to investigate that!

Aircoookie, 2020-12-07:
@Crosstuga I suspect your issue is a bit different - after boot there is only a single segment, this is correct. Have you set one of the presets with both segments set up correctly as Boot preset in LED settings? Then WLED should restore the segments for you on reboot.

mascolo74, 2020-12-07:
> @ mascolo74 grazie per la segnalazione! È inteso che il ciclo si spenga quando si accende o si riavvia, tuttavia non dovrebbe impostare il tempo per preset su `0`. Avrò bisogno di indagare su questo!

grazie a te per quello che hai fatto e fai, rimango in attesa di novità, ma purtroppo nel frattempo dovrò tornare alla versione 10
thanks to you for what you did and do, I am waiting for news, but unfortunately in the meantime I will have to go back to version 10

Crosstuga, 2020-12-07:
> @Crosstuga I suspect your issue is a bit different - after boot there is only a single segment, this is correct. Have you set one of the presets with both segments set up correctly as Boot preset in LED settings? Then WLED should restore the segments for you on reboot.

@Aircoookie You are right on the behavior on the boot but is not linked to reboot just mislead me. Is not the same issue as reported here.
It is the preset cycle when run the cycle segment 2 does not work. If i press button work fine. I just saw the issue https://github.com/Aircoookie/WLED/issues/1447 that describes what I have better and it is confirmed. All good I will wait for fix.
thanks again for your help!

exeljb, 2020-12-12:
I too am having the issue that @mascolo74 is having.  I'm using wled 0.11 to animate some blinking/winking led light bulbs which work great...until they lose power or reboot. then they run through the presets at lightning speeds when the time per preset goes to zero

brownbloodspider, 2020-12-14:
> @mascolo74 thank you for the report! It is intended that the cycle turns off when toggling the power or rebooting, however it should not set the time per preset to `0`. I will need to investigate that!

I have the exact same issue. The time per preset value resets to 0 after a boot. I have only 1 segment and only 2 effects in favorite. Any quick help as this is important for my christmas lights

closed 2021-06-16 by stale[bot]