wled/WLED #1447: Segments >0 not correctly configuredwhen preset  cycle is active

lucaf86, 2020-12-07:
**Describe the bug**
I have a preset with two segment, each segment has a different configuration with diferent effects.
Here the API command: {"on":true,"bri":28,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":298,"grp":1,"spc":1,"on":true,"bri":255,"col":[[0,0,255],[8,255,0],[0,0,0]],"fx":96,"sx":89,"ix":29,"pal":48,"sel":true,"rev":false,"mi":false},{"id":1,"start":1,"stop":298,"grp":1,"spc":1,"on":true,"bri":255,"col":[[0,0,189],[0,0,207],[255,255,255]],"fx":28,"sx":128,"ix":128,"pal":3,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
If I apply the preset standalone it works as expected but if it's part o a preset cycle the second segment has the same configuration of the first segment.

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

lucaf86, 2020-12-07:
I Think I've found the issue but I'm not sure how to fix it.
When the preset is changed inside a preset cycle the function colorUpdated(int callMode) is called and strip.applyToAllSelected = true forces to update all the segments with the same configuration.
Inside setEffectConfig() (called by colorUpdated()) there is a loop which overwrites all the segments:
if (applyToAllSelected) {
    for (uint8_t i = 0; i < MAX_NUM_SEGMENTS; i++)
    {
      if (_segments[i].isSelected())
      {
        _segments[i].speed = s;
        _segments[i].intensity = in;
        _segments[i].palette = p;
        setMode(i, m);
        applied = true;
      }
    }
  } 

maybe is sufficient to avoid forcing strip.applyToAllSelected = true in this case but I don't know if it has consequences.
Has a temporary workaroung, setting "sel":false for the second segments solve the problem

Aircoookie, 2020-12-07:
Thank you very much for the troubleshooting, it is highly appreciated! Will change that right away!
`strip.applyToAllSelected` must indeed be `false` with 0.11 presets - I missed changing this, as `true` was correct in 0.10.

Aircoookie, 2020-12-10:
This should now be fixed in latest master :)

closed 2020-12-10 by Aircoookie