wled/WLED #1453: Error 12: The requested preset does not exist

malexandrum, 2020-12-08:
**Describe the bug**
Freshly written NodeMCU with 4MB flash after `erase_flash`.
After adding 2 presets, this error keeps poping up and preset cycle settings (start, end and duration) do not save:
**Error 12: The requested preset does not exist**

[image]

**To Reproduce**
WLED 0.11 just installed on NodeMCU with 4MB flash.
Configure # of leds to 524
Add these two presets:
{"on":true,"bri":121,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":524,"grp":1,"spc":6,"on":true,"bri":255,"col":[[255,186,121],[255,204,155],[0,0,0]],"fx":78,"sx":143,"ix":230,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

{"on":true,"bri":186,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":524,"grp":1,"spc":6,"on":true,"bri":255,"col":[[255,0,0],[8,255,0],[0,0,0]],"fx":93,"sx":135,"ix":183,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

**Expected behavior**
This error shows up and keeps showing up after reboots and page changes:
**Error 12: The requested preset does not exist**

**WLED version**
 - Board: NodeMCU
 - Version 0.11.0
 - Format Binary

Thank you for your help!

gsieben, 2020-12-08:
I have reproduced the "error". However, this is not a bug. But two necessary features would be helpful.

You have as last preset the number 11, but you only have 2 presets. Presets 3-11 do not exist. Hence the error message. So there should only be the number 2.  Now we come to the missing features. 

1: The "Last preset" value is only saved when "Preset Cycle" is switched off and then on again. In this case a save button or an automatic save would be useful. 

2: If there are only 2 presets, "Last Preset" should automatically go to 2. If there is only one Preset, the Cycle should be switched off automatically.

pbolduc, 2020-12-10:
This is related somewhat related to #1458.

closed 2021-06-16 by stale[bot]