wled/WLED #1456: [DXM/E131] Is channels per universe hard coded in WLED to 510?

timdonovanuk, 2020-12-08:
I've been trying to test multiple universes on one small 50 LED strip, so set up my controller to have 2 universes with each having 30 channels (i.e. 10 RGB LEDs) in XLights:

[image]

I then setup two lines, each with 10 LEDs. The second one starts on channel 31, i.e. universe 2:

[image]

However, only the first universe ever lights up. My best guess is WLED expects there always to be 510 channels on each universe? If so I will log a feature request to set channels per universe in WLED.

Thanks for reading :)

timdonovanuk, 2020-12-08:
Just found this: https://wled.discourse.group/t/missing-171st-pixel-on-wled-0-10-2-with-more-than-one-e1-31-universe/853/10 so I guess that answers my question. I there any way to change the label to a feature request? :)

Aircoookie, 2020-12-08:
Hi, thank you! :)
Is there any usecase for a configurable universe size though? Sending more small universes instead of a few with max. size will lead to a considerable raise in network overhead, so I don't really think it would make any practical sense.

pbolduc, 2020-12-10:
E1.31 uses UDP. Using a lot of small packets could lead to issues. In UDP, there is no guarantee of delivery, ordering, or duplicate protection. A UPD packet contains 20 bytes of header + data.  

https://en.wikipedia.org/wiki/User_Datagram_Protocol
[image]

[E1.31](https://www.diychristmas.org/wiki/index.php?title=Protocols) has over 100 bytes of header for on each packet.

From a network perspective this adds a lot of overhead.

From xLights side of things, is there any reason string 2 cant be on universe 1, starting on channel 31? I have installed xLights but haven't started trying to use it yet.

pbolduc, 2020-12-10:
In the short term, I assume we could change this line https://github.com/Aircoookie/WLED/blob/b5232886d992090bafd719cc74643afa505c1018/wled00/e131.cpp#L3

like

```c++
#ifndef MAX_3_CH_LEDS_PER_UNIVERSE
#define MAX_3_CH_LEDS_PER_UNIVERSE 170
#endif
```

which would allow you to create your own custom build with the number of LEDs / universe.  It is always a can of worms making things configurable, place to edit, logic to save, API logic to edit, logic to load, etc.