wled/WLED #1457: RGBW auto calculation of white channel option returns to Dual after power loss or restart

srdjanv, 2020-12-08:
RGBW auto calculation of white channel option returns to Dual after power loss or restart

**If you go to config > Led Preferences and check 4-channel type (RGBW) option and put it to Accurate like in picture below, save the settings.**

[image]

**now if you reboot or restart device that Accurate option is returning to Dual** 

[image]

**Expected behavior**
On older version this was not a problem 
This is probably connected with new color temperature option.
I use RGBW Cold White strip

**WLED version**
 - Board: Wemos D1 mini
 - Version: v0.11.0
 - Format tried Binary and self-compiled (same issue)

Thank you!

southseaboy, 2020-12-13:
I am also seeing the same issue with 0.11 on ESP32 - Wemos D1 style (compiled with my own Usermod for touch controls).

closed 2020-12-13 by Aircoookie

Aircoookie, 2020-12-13:
Thank you very much - fixed :)

southseaboy, 2020-12-13:
Thankyou - marvellous firmware and excellent support.  Still watching the Home Assistant conference ... interesting new capabilities...

srdjanv, 2020-12-14:
@Aircoookie , Thank you soo much for fixing this! :)

commit b1a7fe0f (2020-12-13): Fixed RGBW mode not saved between reboots (fixes #1457)