wled/WLED #1458: Can't save config with Preset Cycle on AND a Time Per Preset less than 10 seconds

Shoopity1, 2020-12-08:
**Describe the bug**
When turning on cycling through presets, setting a per-preset time of less than 10 seconds, and saving that so it'll cycle at every boot, at the next boot the per-preset time will be 0 and it'll fly through all the presets as fast as it can.

**To Reproduce**
1. Set Time Per Preset to anything less than 10 and turn on Preset cycle.
2. (As per the instructions to set cycling presets on bootup) Go into LED Preferences, check the box next to "Set current preset cycle setting as boot default:" and click save.
3. Reboot and experience seizure.

**Expected behavior**
The desired Time per preset setting to be save correctly.

**WLED version**
 - Board: ESP32 dev
 - Version: 0.11.0
 - Format: Official binary from Github

**Additional context**
Something about how either the cfg.json is written, or how the variables are computed.  E.G. in the preferences.json, putting
"transition":2,
will translate into a 0.2 second transition for the effect (apparently we now have per-preset transitions, awesome).  However, in the cfg.json, putting
"cy":{"on":true,"range":[1,6],"dur":1}},
results in a Time per preset of 10 seconds.  If less than 10 seconds is define, that "dur" variable gets set to 0

Side note (I don't feel this warrants a separate bug, especially since I didn't troubleshoot this at all): At some point in my trying to figure out what was going on, the Universe for the sACN setting got changed from 3 to the default 1.  It could've been me though as I was copy/pasting old json code to see what was changing.

As always, keep up the amazing work.

pbolduc, 2020-12-09:
https://github.com/Aircoookie/WLED/wiki/JSON-API

**dur** - **Array** of time each preset should be kept, in **tenths** of seconds. If only one integer is supplied, all presets will be kept for that time. Defaults to 10 seconds if not provided.

Can you double check your settings

pbolduc, 2020-12-09:
I also notice on a clean install with the flash erased, the UI defaults to first preset: 1, last preset 5. Even though there are no presets.  Once you use the `Set current preset cycle setting as boot default` in the settings/leds page, you get the following error

[image]

mascolo74, 2020-12-09:
I am stepping up to the problem in fact I encounter the same problem, even if I set more than 10, while on esp 8266 d1 mini at restart it resets with boot boot preset 0 not keeping the boot with the preset presets

Shoopity, 2020-12-09:
This is the creator of the issue; I've delete the account that created this and changed the information on my original github account.
There is definitely some shenanigans going on.  I wasn't doing truly good troubleshooting as I was focused more on just getting it to work.  That is to say, I was changing so many things, so fast, and sometimes multiple at a time, that I'm not sure what it was that made me start looking at json files.  And, I didn't experiment with doing a fresh flash.  I only put the steps I did to reproduce.
But I'm glad others are seeing this behavior.

cloggedDrain, 2020-12-10:
I'm having the same problem. Using a D1 mini, I configured 11 presets with an adjusted time per preset value.  I then enabled the 'Set current preset cycle setting as boot default' option.  When the controller reboots, the time per preset value is 0.

closed 2020-12-10 by Aircoookie

Aircoookie, 2020-12-10:
Fixed in master now!

commit b5232886 (2020-12-10): Fixed preset cycle default duration rounded down to nearest 10sec interval (fixes #1458)

Enabled E1.31/DDP/Art-Net in AP mode