wled/WLED #1460: Sync the segments settings.

WerkAG, 2020-12-08:
In first congratulations for this great project.

I built two equal lamps, and I would like to sync the effects programed on unit 1 to unit 2. This works really well, except if on Unit 1 I program an effect which, per example uses the Spacing setting. The Spacing setting isn't passed to Unit 2, so the effect results different on both units.

I'm starting with WLED so I'm not sure if this is an issue, or is just as it should be. A brief explanation wil be appreciated.
Thanks

pbolduc, 2020-12-10:
You can use either enter the segments manually, or use the [HTTP-Request API](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API#segments) or [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API)

1. On light 1: to a get with your browser: http://_ip1_/json/state  You can remove parts that you dont want like `"udpn": { ... }`
1. Post the **seg**ment part to the second WLED instance back to http://_ip2_/json/state

To post the data you can use a number of tools such as postman, curl, node-red, etc

Aircoookie, 2020-12-10:
Good point on the API, though I will transform this into a feature request about a future possibility of multi-segment UDP sync.
I don't yet know how I will do it since different lamps usually have different LED counts and setups and thus the segments won't be mappable 1:1 in most cases.
At other times, you might actually want the current behavior and only sync a single segment. This means this feature would need to be highly customizabel so you could set what to sync and what not to sync. Will think about how to best add that :)

Aircoookie, 2020-12-10:
In the meantime, it might be feasible to additionally sync the grouping and spacing of the main segment.

WerkAG, 2020-12-14:
Thank you Aircoookie for having considered my question for a future enhancement.
In truth I was thinking in decorative lamps all them equal with the same number of LED, where a 1:1 mapping could happen, but seems you are thinking in doing it even better, and turning that feature it in something "highly customizable "

> 
> 
> In the meantime, it might be feasible to additionally sync the grouping and spacing of the main segment.

For my present case, this would be enough

blaab2, 2021-09-06:
As @Aircoookie pointed out, there might be different lamps with different LED counts and therefore syncing might be an issue. However, I suggest to sync with other lamps segment wise. So color and effect of every segment0 is synced over all lamps. Color and effect of every segment1 is synced over all lamps and so on. What do you think?

Aircoookie, 2022-04-16:
Implemented as of v0.13

closed 2022-04-16 by Aircoookie

PacoSwifty, 2022-05-06:
Hey @Aircoookie, I've been playing with WLED for the past few days and it's just awesome, so thanks for creating it. One of the things I'm trying to wrap my head around pertains to some of the discussion in this thread as far as segments, presets, and syncing goes. It's possible that you addressed this with v0.13 and I'm just not understanding how to configure it properly, but perhaps you could share some insights:

Within one instance I can create different presets with different numbers of segments. For example, say I have two light strips with 50 LEDs each. For Preset A, I have a single segment of 100 LEDs and Preset B I have two segments with 50 LEDs each. As I toggle between my presets I see the segments get removed/reinstated correctly.

Now let's say I introduce another instance running on a separate ESP board but with an identical LED setup - two strips of 50 LEDs each - and I enable syncing between them. As I enable presets in one instance I can see the other synced instance attempt to apply the effect, but it does not inherit the segment setup. For example, let's say Instance B currently has a solid color applied to a single, 100 LED segment. In Instance A, I select a preset that applies the drip effect to two 50 LED segments. Back on Instance B I can see that it has changed from a solid color to the drip effect, but it is applying the drip across its single 100 LED segment rather than inheriting the new configuration of two 50 LED segments. I know above it was stated that this logic is tricky because not being able to guarantee a 1:1 mapping means that things could get out of sync pretty quickly, but I'm curious what your solution was in 0.13 - is it possible to achieve what I'm asking about? I can provide picture/video/better explanation if necessary.

blazoncek, 2022-05-06:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
The issue is closed.