wled/WLED #1461: Expose function to determine if the strips are receiving updates

pbolduc, 2020-12-08:
**Is your feature request related to a problem? Please describe.**
When a usermod is performing actions, like reading a one wire sensor, it can impact the updating of the LEDs. This is most prominent on ESP32 where the LEDs are updated asynchronously using the [RMT method](https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods#neoesp32rmt-methods). There is already a global variable `_lastShow` that captures the `millis()` just after `bus->Show()` is called. In asynchronous methods this `_lastShow` has the time the update started. In synchronous methods `_lastShow` has the time the update completed.

**Describe the solution you'd like**
I would like to have a function that returns a Boolean value indicating if any strip is being updated at the time of the call. The NeoPixelBus library exposes the function [bool CanShow()](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object-API#bool-canshow)

> If the method for the defined bus is hardware that sends asynchronously, then call CanShow() will let you know if it has finished sending the data from the last Show().

In single strip version, the function would something like,

```C++
bool WS2812FX::getIsUpdating(void) {
  return !bus->CanShow();
}
```

In future multi-strip versions, it return return true if any strip is being updated.

```C++
bool BusManager::getIsUpdating(void) {
    for (int i = 0; i < numBusses; i++) {
       if (!busses[i].CanShow()) {
         return true;
       }
    }
    return false;
}
```

**Describe alternatives you've considered**
none

If the approach is accepted, I can submit a PR. Please advise if you have a preference on the function name.

Tasks:
1) Add `CanShow()` function to `NeoPixelWrapper` class to return the value from `NeoPixelBrightnessBus` instances. Since this class is a wrapper, it makes sense to use the same function name.

[image]

2) Add function `getIsUpdating()` to WS2812FX as shown above. Since this is what WLED interacts with, I propose a logical name. I actually prefer `bool updating()`.

pbolduc, 2020-12-08:
The issue is related to this bug I created on my fork: https://github.com/pbolduc/WLED/issues/1

Aircoookie, 2020-12-10:
Implemented by your PR #1366 :)

closed 2020-12-10 by Aircoookie