wled/WLED #1463: MAX_LEDS 1500 has front-end limitation?

roelbroersma, 2020-12-09:
I've compiled WLED 0.11 (latest source code) with  MAX_LEDS 2400 in my_config.h  (also removed the # and space)
It still was having a limit of 1500 in the Config web-interface.

Then I re-compiled and also changed the const.h to #define MAX_LEDS 2400

Still when in the CONFIG web-interface the max is 1500 and when I press the up button it won't go higher.
When I manually type 2400 and click SAVE it says: "Value has to be lower or equal than 1500".

Is there a hard limit (accidentally) in the front-end?

Board:  [ESP32 dev with multistrip mod]
Hardware: LilyGO TTGO T7 Mini32 V1.4
Version [0.11]
Format [self-compiled]

PS. I found (after some research) the issue might be in   html_settings.h  (do a search for 1500 there).

pbolduc, 2020-12-09:
You will need to change the web page source here https://github.com/Aircoookie/WLED/blob/7684fb83d6b6c9dc0afd0131bdc5374333e03bad/wled00/data/settings_leds.htm#L97

Once you change the htm page, you need to run the tools/cdata.js script to update the header file that gets the htm pages embedded.

pbolduc, 2020-12-10:
FYI, At 2400 pixels, you will be limited to about max 13.8 frames/second updates. Some effects may not look good. WLED tries to do 42 frames/sec.

Doyle4, 2020-12-11:
Sorry to interrupt, can we change fps or is it fixed?

pbolduc, 2020-12-11:
Little off topic, but there are two things you can change to have FPS change:

https://github.com/Aircoookie/WLED/blob/b5232886d992090bafd719cc74643afa505c1018/wled00/FX.h#L56

and 

https://github.com/Aircoookie/WLED/blob/b5232886d992090bafd719cc74643afa505c1018/wled00/FX.h#L75

Doyle4, 2020-12-11:
Thank you and sorry again for jumping into someones issues.

closed 2021-06-16 by stale[bot]