wled/WLED #1467: Using AUX PIN or single channel PWM on esp32_multistrip?

poldim, 2020-12-10:
**Describe the bug**
I've got a mosfet connected to a pin that's controlling analog led brightness.  Ideally, I'd like to control the brightness of this one LED with monochrome effects or just brightness control (pwm adjustment).  On my way to trying to get this to work, I've found that setting 

**To Reproduce**
Configure the aux pin on v.11

```
#ifndef AUXPIN
  #define AUXPIN 23  //debug auxiliary output pin (-1 to disable)
#endif
```

**Expected behavior**
Successful compile

**WLED version**
 - Board: ESP32 dev
 - Version: 0.11.0
 - Format: self-compiled, esp32_multistrip

Thank you for your help!

Aircoookie, 2020-12-10:
Hi! 
`AUXPIN` was added years ago for some kind of debug functionality I can't even remember anymore. I might consider removing it. In any case, it won't do anything you'd expect it to do as IIRC it can only be triggered via the API.

If you want PWM control of standard LEDs using a mosfet, un-comment `WLED_USE_ANALOG_LEDS` and set your pin(s) starting in line 100 of `NpbWrapper`. Edit: I just realized both the ESP32_multistrip usermod and the multistrip fork by peacepenguin have removed the support for analog LEDs :( so for now it is only possible if you run the standard `NpbWrapper.h` with a single digital output. Running multiple digital busses + PWM will be supported in 0.12 :)

poldim, 2020-12-10:
Thanks for the quick response! 

> `AUXPIN` was added years ago for some kind of debug functionality I can't even remember anymore. I might consider removing it. In any case, it won't do anything you'd expect it to do as IIRC it can only be triggered via the API.

I was assuming the `AUXPIN` (output) was similar to the `RLYPIN` but would allow control from a separate button/api call.  This way it wouldn't be tied to the state LED power and could be controlled separately.

> If you want PWM control of standard LEDs using a mosfet, un-comment `WLED_USE_ANALOG_LEDS` and set your pin(s) starting in line 100 of `NpbWrapper`. Edit: I just realized both the ESP32_multistrip [...]

Would the standard `NpbWrapper.h` with a single digital output support having a single digital string and simultaneous but separate control of an analog PWM channel (using WLED_USE_ANALOG_LEDS) on a second pin or would that essentially make it a _multistrip_?  

I'm trying to setup multiple strings of WS2811s for the bulk of the christmas tree and then have an analog LED channel for controlling the star at the top of the tree.  If need be I can wire the multiple strings in series to make one long strip and just split them up via segments, but the analog PWM might have to wait.  

_As an aside:_ Do you have any plans to support PWM channels that are not tied to r/g/b?  Specifically, I'm thinking of something like the 5CH H801, but connecting 5x strips that are just a single color white for something like kitchen cabinet lighting.

Aircoookie, 2020-12-10:
> I was assuming the AUXPIN (output) was similar to the RLYPIN but would allow control from a separate button/api call. This way it wouldn't be tied to the state LED power and could be controlled separately.

It is. But by default it is set to turn off after 500ms after the API call, I believe I used it to trigger an relay for automated blinds. Doesn't support PWM though.

> Would the standard `NpbWrapper.h` with a single digital output support having a single digital string and simultaneous but separate control of an analog PWM channel (using WLED_USE_ANALOG_LEDS) on a second pin or would that essentially make it a multistrip?

It is a bit of a hack, currently the PWM LEDs will always be set to the same color as the first digital pixel.

> As an aside: Do you have any plans to support PWM channels that are not tied to r/g/b? Specifically, I'm thinking of something like the 5CH H801, but connecting 5x strips that are just a single color white for something like kitchen cabinet lighting.

Yes 😉

poldim, 2020-12-10:
Awesome, thanks, I'm looking forward to .12!

closed 2021-06-04 by poldim