wled/WLED #1468: Visual Studio Code Compile

Doyle4, 2020-12-10:
I know how to build Marlin firmware for 3D Printers, but WLED is baffling me!

It says (comment this out with a ';' when building for your own board), but if I remove ; on all other boards, build fails, if I leave them on it starts to compile but fails at the end. How do I select the board?

Is there a VSC tutorial anywhere? Id like to build my own with the slider fix.

Doyle4, 2020-12-10:
Think I got it sorry, will close If this build works :)

Doyle4, 2020-12-10:
Failed.

Could someone be so kind and post what I need to do to build a D1 Mini - Pin 4 please.

Doyle4, 2020-12-10:
SOLVED!

closed 2020-12-10 by Doyle4

Aircoookie, 2020-12-10:
Hi, glad you got it worked out!
If somebody else has troubles compiling in VS Code, here is the wiki page on it:
https://github.com/Aircoookie/WLED/wiki/Compiling-WLED#compilation-guide-platformio

Doyle4, 2020-12-10:
Ah! that was the link I have been looking for, will have a good read of that thanks!

I managed to get the D1 to compile, but getting the NODEMCU to compile with custom pin fails constantly, but will have a read before asking anything.

I did spot though, my_config needs to be uncommented wether you are using it or not as it fails if its not uncommented. Again.. I need to read the link :)