wled/WLED #1469: Custom ESP32 Build red led constant blink.

Doyle4, 2020-12-10:
I have managed to compile a D1 Mini firmware all good, and works on the NODEMCU

I changed environment to custom32_LEDPIN_16, Compiled and flashed, the red power indicator LED blinks constantly, I erased its memory, flashed the 32 bootloader and reflashed firmware, LED back to blinking, What am I missing?

I read the VSC link, but to be honest its not a very clear tutorial for someone new like my self sorry. I feel you need to already know what you are doing to follow it, a good breakdown tutorial would be amazing, id also be happy to do one once I have figured it out as I did for others with Marlin firmware for the 3D Printing community.

Not moaning, please don't take this the wrong way.

Doyle4, 2020-12-10:
[image]
[image]

Aircoookie, 2020-12-10:
Maybe the LED blinking is actually fine. Is your WLED-AP reachable? If not you can click the little power plug symbol in the bottom blue bar of VS code to open the serial terminal and see if the board is bootlooping.

Doyle4, 2020-12-10:
Its like the board is bricked, no WLED SSD shows up in wifi settings on my mac..

Doyle4, 2020-12-10:
I have tried flashing back over 0.11.0 that I had previously downloaded (not compiled by me) and thats not fixed it,

Doyle4, 2020-12-10:
serial is showing nothing connected

Doyle4, 2020-12-10:
@Aircoookie Dude... Im so sorry! I was using as ESP8266 flasher, so sorry! using an updated version of the same software fixed the issue using your compiled version, will test mine now.

Again, im sorry.

closed 2020-12-10 by Doyle4

Doyle4, 2020-12-10:
UPDATE:
Working: Own compiled version for D1 Mini and ESP32! Thanks for the slider fix and putting up with me.