wled/WLED #1470: Wrong time with timezone CET/CEST

watho, 2020-12-10:
**Describe the bug**
Using timezone CET/CEST the local time shown is off by one hour.

**To Reproduce**
Set the timezone to CET/CEST and local time is UTC+2

**Expected behavior**
Local time should be UTC+1

**WLED version**
 - Board: Wemos D1
 - Version 0.11.0
 - Format Binary

**Additional context**
My UTC time is correct, so it is a problem with the offset.

Workaround is to set the timezone to UTC and utc offset to 3600 seconds. This needs adjustment every DST change.

Thank you for your help!

Aircoookie, 2020-12-10:
Hi!
I live in Germany and use CET/CEST, too. For me it works correctly. Currently it is CET/winter standard time, and for me local time is UTC+1 as expected.
What is your country? Feel free to send me a screenshot of your time settings, maybe something is wrong...

watho, 2020-12-10:
I am in Germany too. 
[image]

Aircoookie, 2020-12-10:
Sorry, I'm unable to reproduce the issue at all, even with the exact same settings you are using.  
Could I have a look at your `http://[WLED-IP]/cfg.json` file? Don't worry, it does not contain your wifi passwords.

Aircoookie, 2020-12-10:
It might be possible that it was a bad flash somehow... You could try (in that order) to reboot the device, reset settings to factory defaults in Security settings, or erasing the entire flash and re-flashing the binary.

watho, 2020-12-10:
Link to config: https://gist.github.com/watho/f4c43673e42cd7d3384c83f8432a1ef7

Aircoookie, 2020-12-10:
Thank you very much. Everything correct with that, I have no clue what sorcery is going on here :/ Do you have a 2nd board where you could try if it has the issue as well?

watho, 2020-12-10:
The issue is gone. Unfortunately i did two things and can't say if any of those fixed it.

Using the http request api i have set the utc time to 0 = 01.01.1970 00:00:00
I have changed the ntp server to de.pool.ntp.org.

After next reboot the time was correct.

Thanks for this awesome work.

closed 2020-12-10 by watho