wled/WLED #1471: Compatible Hardware

fisheye62, 2020-12-10:
I have some point style led modules I purchased because their operating temperature was good for our climate here in northern Canada (-40C). I have since noticed that the Driver that was used isn't in the the compatible hardware list. Could this driver be added to the list? 
The Driver is UCS2904B for RGBW type

pbolduc, 2020-12-10:
The low level communication is provided by [NeoPixelBus](https://github.com/Makuna/NeoPixelBus). They have a list of [supported pixels](https://github.com/Makuna/NeoPixelBus/wiki#supported-pixels). This [listing](https://www.gree-leds.com/rgbw-digital-strip/ucs2904-digital-rgbw.html) shows a UCS2904 strip and it says it has 5050 RGBW LEDs.  If that strip kind is supported by NeoPixelBus, then it could be supported by WLED.  I would suggest to see if you can find out if your strip kind is supported on NeoPixelBus.

I does make it difficult for the developers to add support to hardware we dont have since there is no way to test.

fisheye62, 2020-12-10:
Thanks Phil
My lights are 12v RGBW UCS2904B 5050 SMD 3-2800 k. 

Kinda stuck where to go from here

 Sent from my iPhone
Tom Dudley
780-914-7089

Aircoookie, 2021-09-11:
If UCS2904B worked for you with the SK6812 profile, it will also work in WLED :)
Please reopen in case UCS2904B still does not work.

closed 2021-09-11 by Aircoookie

alax, 2021-12-30:
I was not able to make UCS2904B LEDs work with the SK6812 profile. I've seen a lot of documentation saying they are similar protocols, but no luck on my end.

al3x1337, 2021-12-30:
I've having no issues with UCS2904B with the SK6812 RGBW - RGB preference.
Originally I had flashing rainbow colours because the ground between my
nodemcu and led strip was not connected but no issues since.

What are you experiencing?

alax, 2021-12-30:
@al3x1337 thanks for the update-- that gives me hope that I'm just doing something wrong. I can't seem to get my lights to do anything at all. I'm using a 5v level shifter on an ESP32, and I can control other (non UCS) light strips with this same circuit.

Since you've confirmed it works for you, I'm going to give it another shot and I'll update here if I make any progress.

alax, 2021-12-30:
Alright, I got it working as well. The issue was wiring on my end, my voltage level shifter was connected to 3.3v, not 5v for power as required.

Although, my white channel goes a little nuts when adjusting colors or setting patterns, it randomly blinks on and off. That might just be my LEDs, but I thought I'd mention it in case anyone else sees similar behavior.