wled/WLED #1474: Update LED / Sync Settings via JSON API

jfrux, 2020-12-11:
I may end up just doing a PR myself but figured I'd post this here first if there is any concerns with doing this type of thing before I attempt it.

I've been managing / calculating my universes, brightness mAh limiter, etc. using airtable and would like to be able to quickly do an HTTP POST to quickly update the power settings and universes as my setup / configuration changes.

I did attempt to do a POST to `/json/info` but it seems to be read only for now.

The things I think I'd like to update or make writable would be the name, led count, and power limit.
Power limit would be nice since I'm dividing the load across my modules connected to a shared power supply.

huggy-d1, 2020-12-14:
POST to /json per json api wiki

blazoncek, 2023-05-27:
Configuration changes can be POSTed to `/json/cfg`.

closed 2023-05-27 by blazoncek