wled/WLED #1475: WLED Sync - mirror effect

Offpiste84, 2020-12-11:
Add mirror effect to Sync WLED devices (UDP Notifier).

Aircoookie, 2020-12-13:
Good idea to sync this - along with grouping and spacing maybe.

tkerby, 2020-12-14:
I'm thinking sync could do with a few updates.  I'd like to be able to choose which other instances I sync with and I'd also like to be able to select clone or extend as an option. Clone copies the effect onto the other instance exactly. Extend should create virtual LEDs on the end of the string - e.g. if you wanted to wrap a building you may need additional ESP8266 boards but you want the effect to jump the gaps

huggy-d1, 2020-12-14:
Ahh a WLED cluster! What a concept!

costyn, 2020-12-17:
This is also a feature I would love. Multiple controllers and have the effect wrap/jump from controller to controller.

huggy-d1, 2020-12-17:
@costyn That is available if external software like xLights is used.  Then it can address all controllers and control at the pixel level across all pixels.

That said, I am a fan of having WLED's UDP sync get new features.  Only problem is they would need to get incorporated into the json API, http api, mqtt api, documentation, UI, and so on.

Like when sync, perhaps they could call it a "join", and WLED act just like a pixel in an addressable LED strip, see the command, act on it if the command is for that WLED strip, and then to pass commands on to the next strip as needed.

It sounds quite challenging until someone can gracefully state a low-processing-overhead low-communication bandwidth protocol that would work.  These I.T. / C.S. people, they like challenges though.

costyn, 2020-12-17:
Thanks, I'll check out xLights. That said, the low level networking already exists for ESP32, like for example Espressifs Mesh https://www.espressif.com/en/products/sdks/esp-wifi-mesh/overview  and there's also painlessMesh https://gitlab.com/painlessMesh  

I have made a successful integration of painlessMesh with FastLED. The patterns are exactly matched, it's pretty cool. https://github.com/costyn/LEDswarm. But the hard part is spanning 1 pattern over multiple mesh nodes. I was hoping someone else was maybe already working on it :)

blark, 2021-12-03:
+1 for extra sync options such as mirror and effect wrap/coordination. I have about 6 strips on the soffits of my home that it is impractical to run with the same controller. Effects such as scanning seamlessly transitioning between WLED would be awesome.

blazoncek, 2021-12-03:
You can use virtual strips since 0.13-b4.

blark, 2021-12-03:
@blazoncek very cool thanks, I'm running b5 I will check it out tonight to see if I can figure that out.

blazoncek, 2021-12-17:
Working on a fix for OP.

blazoncek, 2021-12-17:
Fix available for testing in my fork, sync-seg branch.

closed 2022-01-25 by blazoncek