wled/WLED #1476: NodeMCU/WLED Fails above 5000 mA

willi3by, 2020-12-12:
**Describe the bug**
I currently have 3 5 meter 60led/m strips of WS2812B LED lights, rated at 0.3W per light. I plan on adding two more strands. For the project I bought a 5V 30A power supply that is connected at the beginning of the lights and a nodemcu esp8266 connected directly adjacent to the first led. In WLED, I initially set the current to 3000 mA and things work as expected, but the lights are not very bright. When I try to turn up the amps, some of the patterns start failing. Anything above that, and nothing works, the lights are just bright white for the first couple of LEDs and a gradient of red along the rest. My current set up is 900 LEDs. When I say fail, I mean that I lose connection with the nodeMCU and have to unplug and re-flash it to get it working again. I am not sure what is causing the error, whether it is with my setup or the app. 

**To Reproduce**
Turn amps to greater than or equal to 5000 mA. 

**Expected behavior**
I expect the lights to keep working as expected all the way up to the rating on my power supply. 

**WLED version**
 - Board: ESP8266MOD
 - Version: 0.11.0
 - Format: Binary

**Additional context**
Any help is greatly appreciated. This is my first LED project so I am sorry if there is something obvious I am missing.

Thank you for your help!

bgutman, 2020-12-12:
Need to know a little more: 

- Do you have any [power injection](https://github.com/Aircoookie/WLED/wiki/FAQ#my-leds-dont-get-as-bright-as-they-should) between your three led stripes?
- How do you have the esp8266 connected to the strips, directly or mounted onto a [boards/shield/addon](https://github.com/Aircoookie/WLED/wiki/Compatible-hardware) or level shifter ?
- Have you read the [wiki](https://github.com/Aircoookie/WLED/wiki), WLED supports 1500 but recommends [750 leds](https://github.com/Aircoookie/WLED/wiki/FAQ#my-device-is-unresponsive-or-animations-lag)?

willi3by, 2020-12-12:
Hi,

Thanks for the response! 

I don't currently have any power injection anywhere except the beginning of the first strip. I definitely see a difference in brightness between strip 1 and 2 (not so much 2 and 3), so I am going to add a parallel injection line, but I didn't think that would cause the issue. 

It is connected directly to the board. I thought about addons but it seems like the data is traveling ok since the effects work at lower amperage. 

That's interesting! I haven't read through most of the wiki yet but definitely will look through it. I was actually happy with the performance at 900, just wanted extra brightness so I cranked up the amps and ran in to the issue. I have two more nodeMCUs with WLED flashed so I can always add another point if needed. 

Thanks again for your time!

yetdog, 2020-12-13:
With that level of density (60led/m) at 5v, you're going to have to run power injection between every strip, or you're going to see some pretty significant artifacts of voltage drop. Also:

- Are you using a level shifter or other means of boosting your data wire prior to your first LED strip?
- Isn't the voltage limiter via WLED only if you're also powering your strips via the ESP board itself, and not using an external power supply? Have you tried disabling the brightness/amperage limiter in WLED?
- Some effects are heavier than others, resource-wise. At 900 LEDs (also the size of my setup), it's best to run an ESP32.

Legsmaniac, 2020-12-14:
There isn't a voltage limiter in WLED, it's a current limit.
And my understanding of it is it doesn't control the supply of power to the strips but limits the brightness of the LEDs which in turn limits the current drawn by them.
The device cannot supply more than 1 amp without frying the board so it's not advisable to power strips via the board. You can set a limit up to 10A without a warning (I think it's 10, I could be wrong) but there's no upper limit I believe.

willi3by, 2020-12-14:
@yetdog thanks for the info! Still waiting on the splicers for the power injection but just disabling the current limit on WLED both solved the issue I was having and greatly improved the brightness of strips 2 and 3. Luckily the MCU is within about 8 inches of the first LED so no need for the level shifter and I am powering the board from the lights instead of the other way around. With power injection, I think everything will be working as expected.

closed 2020-12-14 by willi3by