wled/WLED #1477: E1.31 DMX support for 5ch H801 controllers

danielhelmstedt, 2020-12-12:
I'm starting a project and hoping to use WLED via Artnet from lighting consoles. I've compiled an ESP8285_5CH_H801 binary and WLED itself is working fantastic, however only the RGB channels seem to be controllable via DMX.

This is using Single RGB mode. I have also tried Multi RGBW, and still only have control of the RGB, no W.

Is it possible to get a 5ch E1.31 DMX input option?

Thanks for all your amazing work!

CTI-Tim, 2021-05-26:
I also would love this feature.  it would make these devices very useful for  holiday and general lighting

mxklb, 2023-11-08:
Have you tried sending 

- 4 channels in Single RGB Mode (instead of 3)? 4th channel value is W
- 5 channels in Single DRGB Mode (instead of 4)? 5th channel value is W

So I don't know what H801 is, but this shouldn't matter, setting WLED to one of this modes shall already utilize W.

P.S. This is a hidden undocumented feature I've seen in the code. If you can confirm that this is already working, We could add missing docs and this issue can be closed. IMO this is already working, since 4lloyd committed 8b79a97, 19 months ago (April 3rd, 2022 5:52 PM) .. give it a try ..

CTI-Tim, 2023-11-09:
Wait so you are saying that in DMX mode just send extra data and WLED will figure it out?  because I can not find any setting to change from RGB to RGBW.  none of the releases will do what you are claiming is already in place when I send RGBW, the W ends up as the R channel on the next fixture address.  or is this in the working code only and none of the releasses and I need to compile the latest code myself to see the place to turn on or off the W channel?

mxklb, 2023-11-15:
> when I send RGBW, the W ends up as the R channel on the next fixture address ..

this is exactly the undocumented "feature" as implemented. Have you tried using 4 channel DMX fixtures (with RGBW) in your DMX console/app and using _Single RGB mode_ WLED controllers setup to successive DMX start addresses **1**..4, **5**..9, **10**..14, .. If you do so, you shall be able to send W as desired. If so, this issue may be obsolete - already implemented.