wled/WLED #1479: Light Painting Support

tkerby, 2020-12-12:
It would be fantastic to have support for light painting and POV effects by sending image data quickly to an LED strip.

I can see a few ways of operating

1 - Read data from an SD card, select it as an effect in the interface to choose the sequence
2 - Stream data over UDP quickly enough

I don't know what the limitations are in terms of refresh rate - typically the commercial products are selectable refresh up to about 120Hz depending on how fast the person is moving

Vinno97, 2020-12-13:
Have you checked if the current UDP performance is actually not fast enough? WLED works very well with both Hyperion and Prismatic, both Ambilight-like applications where the full LED-strip has to react quickly to a moving image. 
There is also built-in support for E1.31. 

There's a good chance WLED is already fast enough for your purpose (via option 2).

Stefan231278, 2020-12-22:
Hi
I`m new in this, but would like to make some Image-Painting with an LED-Stick. Its controlled by an ESP32. an WLED works very well. 
Which application are the best for lighten up and stream an Image?
I searched in Hyperion, but i didn`t find any solution.
Thx for response.
Stefan