wled/WLED #1482: WLED Ambilight over USB for Dummies

TarFilarek, 2020-12-13:
Hi! 
First of all - thank You for this amazing project! 

I'm trying to connect Prismatik (unofficial) with WLED according to this thread:
https://wled.discourse.group/t/wled-ambilight-hyperion/102

Settings:
Adalight (Serial)
Serial port: COM9 (checked with Device Manager)
Baud rate: 115200
Color format: RGB

Prismatik seems to see the device. When I chose any other COM port, there is a red sign instead of sun/moon icon. 
Still.. it just doesn't seem to have any effect on WLED. Led Strip shines in the previously chosen scheme from WLED app. 

What am I missing?

2nd question - Is it possible to control WLED with Hyperion app over USB (for PC monitor) without any other devices like Raspberry Pi etc.? If yes - could You give me some tips on how can I set this up?

Aircoookie, 2020-12-14:
Hi!
That is very strange, your settings appear correct and that should work.
When you start the adalight output and go to the WLED control UI, is anything different? In particular, it should display a star with the text "WLED is receiving live data".

About your 2nd question, that should be possible too, either via Adalight or TPM2 over USB. Also check out AmbiBox if you are using windows, it works quite well in my testing :)

TarFilarek, 2020-12-15:
Unfortunately no - everything looks the same in WLED UI. I've checked Prismatik, Hyperion and AmbiBox and...

During configuration in Prismatik and AmbiBox there is just a little blink of led strip, it changes its colour to orange (or some other previously launched colour), and that's all. There is a similar reaction when I close the Prismatik or AmbiBox.

Also, each time I end the configuration process in Primatik app, a warning "Connection to light failed!" pops out at the WLED UI and after a second or two everything is back to normal.

closed 2021-06-16 by stale[bot]