wled/WLED #1487: Requesting addition of installed WLED version to be exposed in Home Assistant integration

michaeldvinci, 2020-12-14:
I am trying to create a Home Assistant Entity that will monitor releases of WLED published in the repo vs the one installed on the esp32. Monitoring releases is easy with the 'github:' sensor, but I am hoping to get the WLED integration to expose the installed version on our connected devices.

If the Home Assistant integration also exposed the version installed (i.e. v0.11.0), I could compare the current release to current installed, create a list, then push a notification to my phone. 

Example: I have 5 WLED setups in my place. It'd be nice if I could aggregate a lsit of versions from the current integration and compare them against the most recent release from your repo.

I tried using the python API wrapper in Node-Red to no avail.

DerFlob, 2020-12-14:
I think you'll have to go to [home-assistant/core](https://github.com/frenck/python-wled) for that feature request, since the HA integration is not part of this project, but HA itself. WLED does already expose the version through its JSON API and the [Python-wrapper used by HA](https://github.com/frenck/python-wled) does allow fetching the version as well (see its example). 

HA actually does show the WLED version in the device info of the WLED device, containing the different sensors. I don't know if or how you could use it in your automations though. If not possible, you could use a [REST sensor](https://www.home-assistant.io/integrations/rest/) to retrieve the version from WLED's [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API) manually.

blazoncek, 2022-07-25:
Version and build numbers are exposed in JSON info object avaialble at /json or /json/info

closed 2022-07-25 by blazoncek