wled/WLED #1494: WLED on ESp32 or esp8266  MATRIX 900 LED - Can WLED drive multi Universes?

flixmyswitch, 2020-12-15:
Very new to this but having fun

I have a matrix with 864 LED.  data in from the bottom left and all leds connected in series

I have it running through xlights and the matrix works, but the top 288 leds  , panel flickers strange colours.

I can still see the whole image on the matrix 32 x 27 only this flickering and colour issue.  I have a logic level shifter installed

i am sure i have not the this up correct but have not been able to find any guidance.

How should i set up WLED to control 864 pixels.

In xlights i have i only have 1 universe set up with 6 universe counts at 510 channels per universe.  And i think this may be my problem.

Is anybody able to give me some pointers.  So i think i need how to set up multiple universes with WLED, for xlights i can always ask in their ZOOM room

Other than that. Thanks WLED for getting me started

Thanks

Andy

bgutman, 2020-12-16:
Do you have any power injection between each panels?  Sounds more like voltage drop issues with adding each panel
I would use the esp32 for this.

WLED can handle the large number of pixels and in xlghts your device will span multiple universes.

flixmyswitch, 2020-12-17:
Thank for the reply

ITS ALL SORTED.  USER ERROR AS NORMAL.

the way i had constructed my matrix in 3 sections.

Each section had its own power and power injection every 50 pixel back to a common power supply which also powers the ESP.

So what i did (WRONGLY) when going the three panels of 288 pixels together was ONLY like the DATA line between the 3 panel.  Yes this worked but gave me strange colours in the top panel.  Having fault tested for two days, cutting out pixel which i thought were bad.  I then found my error.

I should of also linked the negative between the panel, once i did this the real magic started and displays well.

So thanks for your help.

closed 2020-12-17 by flixmyswitch

adamcoulombe, 2020-12-22:
I am curious, How do you configure your matrix in xlights to use multiple universes? I am trying to figure this out right now

flixmyswitch, 2020-12-22:
strings 1
nodes/strings 864 (the number of pixels you have
strands/strings 27   (my matrix is 32x 27 rows = 864 and the rows zig zag left to rigth
I have one esp8266 with wled being pass data from xlights