wled/WLED #1496: ESP8266_1M_ota Version with GPIO14 Data

bepece1, 2020-12-15:
Hi!!
I would like compile this binary "ESP8266_1M_ota.bin" with GPIO14 like DATA LED.

I'm using Sonof Basic R1 and I need use GPIO14 instead of GPIO02

Is it possible?

I don't know like do it :(

Helpppppp

THANK YOU ver much

bepece1, 2020-12-15:
I found a solution

closed 2020-12-15 by bepece1

seblabe, 2020-12-18:
Hi, I am having the same problem. Can you share the solution? Thanks!

bepece1, 2020-12-19:
> Hi, I am having the same problem. Can you share the solution? Thanks!

I installed all the apps, and dependencies, in my computer. Now I can compile the source code in my sonoff basic with gpio14 like data output
If you want I can generate your built and send you, you only must tell me your wifi user/password for coding right the compilation

seblabe, 2020-12-19:
Thanks, would the WIFI data be hard coded or can I change it in case I need it?

bepece1, 2020-12-19:
> Thanks, would the WIFI data be hard coded or can I change it in case I need it?

Yes, you can change it after if you need, but the first time I think you need connect to your wifi you can connect to the device.

bepece1, 2020-12-19:
Here, you have a build 
https://www.transfernow.net/ddl/bepeceWled

User Wifi: bepece
pass Wifi: bepece1234bepece

seblabe, 2020-12-19:
Thanks,

Tried the bin file, but this release doesn't seem to work on my SONOFF. I am able to get WLED_0.9.1_ESP2866_1M_ota.bin to work, however I have the issue that the board has no GPIO2 exposed there.
When I tried your release I can't see the WLED wifi network, any idea why this may be?
Thanks!

https://www.transfernow.net/ddl/bepeceWled

bepece1, 2020-12-19:
> Thanks, Tried the bin file, but this release doesn't seem to work on my SONOFF. I am able to get WLED_0.9.1_ESP2866_1M_ota.bin to work, however I have the issue that the board has no GPIO2 exposed [...]

You must create wifi access point with name "bepece" and password "bepece1234bepece" and your device connect automaticly to your wifi

If you want I can generate your built and send you, you only must tell me your wifi user/password for coding right the compilation

seblabe, 2020-12-19:
Hi,

I tried it with my phone, but I can't see any devices connecting to it. I think it may be an issue with the chipset or the release since if I try release 0.9.1. I can connect to it, with your bin and anything besides that release, the WIFI doesn't seem to work.

Any ideas?

How can I send you my credential securely?

________________________________ From: bepece notifications@github.com<mailto:notifications@github.com> Sent: Saturday, December 19, 2020 11:04:08
AM To: Aircoookie/WLED WLED@noreply.github.com<mailto:WLED@noreply.github.com> Cc: seblabe sblanco@hotmail.com<mailto:

bepece1, 2020-12-19:
> Hi, I tried it with my phone, but I can't see any devices connecting to it. I think it may be an issue with the chipset or the release since if I try release 0.9.1. I can connect to it, with your [...]

bpcmiguel@hotmail.com

FulvioSpelta, 2020-12-21:
Maybe better if you can fork wled and share your work at source level using github.
Mind to hide your private info before publish.

Thanks
f

bepece1, 2020-12-21:
> Maybe better if you can fork wled and share your work at source level using github.
> Mind to hide your private info before publish.
> 
> Thanks
> f

The source code is the same at the original. My problem was install all the apps, and dependencies, in my computer.

FulvioSpelta, 2020-12-21:
> The source code is the same at the original. My problem was install all the apps, and dependencies, in my computer.

I kindly ask you to share the modification / customization of the source in order to be able to compile it (for sonoff basic with GPIO14)  myself. I've already have the vstudio up and running and I'm already able to compile the wled source.
Thanks

bepece1, 2020-12-21:
> I kindly ask you to share the modification / customization of the source in order to be able to compile it (for sonoff basic with GPIO14) myself. I've already have the vstudio up and running and I'm [...]

Change in wled.h
`// You need to choose some of these features to disable:
#define WLED_DISABLE_ALEXA       // saves 11kb
#define WLED_DISABLE_BLYNK       // saves 6kb
#define WLED_DISABLE_CRONIXIE    // saves 3kb
#define WLED_DISABLE_HUESYNC     // saves 4kb
#define WLED_DISABLE_INFRARED    // there is no pin left for this on ESP8266-01, saves 12kb
#ifndef WLED_DISABLE_MQTT
  #define WLED_ENABLE_MQTT         // saves 12kb
#endif
#define WLED_ENABLE_ADALIGHT       // saves 500b only
//#define WLED_ENABLE_DMX          // uses 3.5kb (use LEDPIN other than 2)
#ifndef WLED_DISABLE_WEBSOCKETS
  #define WLED_ENABLE_WEBSOCKETS
#endif

#define WLED_DISABLE_FILESYSTEM        // SPIFFS is not used by any WLED feature yet
//#define WLED_ENABLE_FS_SERVING   // Enable sending html file from SPIFFS before serving progmem version
//#define WLED_ENABLE_FS_EDITOR    // enable /edit page for editing SPIFFS content. Will also be disabled with OTA lock

// to toggle usb serial debug (un)comment the following line
//#define WLED_DEBUG
`
In npbwrapper.h
`//PIN CONFIGURATION
#ifndef LEDPIN
#define LEDPIN 14  //strip pin. Any for ESP32, gpio2 or 3 is recommended for ESP8266 (gpio2/3 are labeled D4/RX on NodeMCU and Wemos)
#endif`

and in your my_config.h your wifi setting.

FulvioSpelta, 2020-12-21:
Thank you very much. I'll try starting from the last version and i'll provide feedback here.

DeviousPenguin, 2020-12-23:
Please let me know how you get on with this, I'd like to use GPIO 14 also

FulvioSpelta, 2020-12-23:
Please be patient I'm going to try this in the next holidays. I'll post here the experience.

bepece1, 2020-12-23:
> Please let me know how you get on with this, I'd like to use GPIO 14 also

https://github.com/Aircoookie/WLED/issues/1496#issuecomment-748892785

FulvioSpelta, 2020-12-26:
Hi all, I've done some work but without any success. I report here, hint are welcome.

**reference hardware**
Sonoff Basic R1

**1) Environment**
I've set-up the vscode and platformio environment following the wiki and i'm able to compile both the
default_envs = travis_esp8266, travis_esp32
using version 0.11.1 source untouched (as-is)

**2) stock firmware**
The idea is that we can use GPIO pin 2 too that is available either in R1,
anycase, due to my little experience with this fw and compiling environment, i'd prefer
at first to have a reference bin version in order to be able to recompile it with success before
to start to make changes.

I've flash the following stock (precompiled) binary
"WLED_0.10.2_ESP01_1M_full_fix2009030"
"WLED_0.11.1_ESP8266_1M"

But i've found the problem described here #461  (posted a feedback too)

**Next steps**
1) I'm waiting for hints about the bootup problem.

2) In the meanwhile I kindly ask for hints and support about:
(referring 0.11.1 version source)
What env to compile to have a fw that is expected to work on sonoff basic R1
How to activate debug (in order to try to find info for troubleshooting)

3) generally
Share a bin or, better, a source that is proven to work and info about
to compile it (with and without debug)

I'm new here so i'm trying to avoid errors and time waste in environment setup
in order to help in troubleshooting.

For completeness I've found this tutorial on youtube
https://www.youtube.com/watch?v=pOW17tnIHeM

Thanks

FulvioSpelta, 2020-12-31:
Hi all, I'm in trouble. In my test only WLED_0.8.6_ESP01_1m_ota.bin seems to work.
Details about my tests here #1590 

@bepece1 can you kindly share the complete source in order to try myself to compile and flash ?
thanks

FulvioSpelta, 2021-01-01:
Hi all, please look at #1590  for final summary.
Thanks

bepece1, 2021-01-02:
> Hi all, I'm in trouble. In my test only WLED_0.8.6_ESP01_1m_ota.bin seems to work.
> Details about my tests here #1590
> 
> @bepece1 can you kindly share the complete source in order to try myself to compile and flash ?
> thanks

Of course.
https://www.transfernow.net/ddl/bepeceWledFlash0-10-2

FulvioSpelta, 2021-01-02:
Thanks, I've downloaded the package for future reference. 
As you can see, in the meanwhile I've compiled a 0.11.1 that's running: [https://github.com/Aircoookie/WLED/issues/1496#issuecomment-753332304](https://github.com/Aircoookie/WLED/issues/1496#issuecomment-753332304)
I'll post further news as soon as i'll be able to build a level shifter and do a complete test.
My idea is to try using both the standard GPIO02 and the GPIO14.

PS
Quickly reading your source i'm not able to understand what env you have compiled and you are using because i don't find the uncommented env in platformio.ini nor i can find any reference to LEDPIN 14

bepece1, 2021-01-02:
> Thanks, I've downloaded the package for future reference.
> As you can see, in the meanwhile I've compiled a 0.11.1 that's running: [#1496 (comment)](https://github.com/Aircoookie/WLED/issues/1496#issuecomment-753332304)
> I'll post further news as soon as i'll be able to build a level shifter and do a complete test.
> My idea is to try using both the standard GPIO02 and the GPIO14.
> 
> PS
> Quickly reading your source i'm not able to understand what env you have compiled and you are using because i don't find the uncommented env in platformio.ini nor i can find any reference to LEDPIN 14

https://github.com/Aircoookie/WLED/issues/1496#issuecomment-748892785