wled/WLED #1498: WLED toggles off lights randomly

mromarreyes, 2020-12-15:
My WLED lights are randomly turning off after about 1 hour of running. WLED is still on because I can access it from the web/app. It seems like the power button in the app gets toggled off. I tried re-flashing a brand new NodeMCU. I tried replacing the PSU. I am using the WS2813 led light strip. Any ideas on what I can try? This led strip has 4 cables. Red, white, green, blue. This is how I have it wired:

LED Red -> NodeMCU Vin
LED White -> NodeMCU GND
LED Green -> NodeMCU D4
LED Blue is not connected (is it required?)

The PSU I'm using: https://www.amazon.com/gp/product/B07Q26YG61/

jimfong1, 2020-12-17:
I have this same problem after upgrading to 0.11.0.  It would turn off after about an hour.  Using a Wemos D1.  I went back to the previous version and no problems.

mromarreyes, 2020-12-17:
I would like to add more of my findings:

1) The timer was disabled so this was not it.
2) Reset all my macros back to defaults (0 0 0). Bingo! This seems to have fixed it for me but I don't understand why. 

I had presets to automatically turn on the lights (T=1) and then turn off the lights (T=0) after a certain time. I will manually turn on and off my lights for now until I can figure out what I did wrong.

jimfong1, 2020-12-17:
I wasn’t using timers or macros. I use a external  AC timer switch instead.   Didn’t have time to play around since I needed the Christmas lights to work.

deltaflyer4747, 2021-01-08:
for me it is caused by ESP resetting, most probably due to power instability. I wrote myself a simple script running on my local linux server every 60s that grabs state of WLED each minute and if it detects it went into specific profile i configured to be used on restart it calls proper preset based on current time of day. So at worst my lights are off for 1 minute.

deltaflyer4747, 2021-06-03:
So i added quite beefy cap right onto esp pads, yet it still resets... So no idea, what's going on.

closed 2022-03-15 by blazoncek

blazoncek, 2022-12-09:
If you are experiencing LEDsturning off every hour, check if you have timed preset with hour set to 24.
24 in hour field means execute preset every hour, not at midnight.

deltaflyer4747, 2022-12-09:
Not a timer issue, happens randomly. I suspect this might be related to hw issue caused by power instability, but i added oh-so-many filtering caps directly to my ESP board, that i can basicaly rule that out...

blazoncek, 2022-12-09:
@deltaflyer4747 Connect USB and monitor serial output. Preferably using debug build.

deltaflyer4747, 2023-01-17:
forgive my stupidness, but ... where can i find debug build?