wled/WLED #1500: Is it possible 6803 IC Chip LED to operate? #851

WerkAG, 2020-12-16:
> 
> 
> Hi,
> it could be that the LPD6803 uses either the WS2801, LPD8806, P9813 or APA102 protocols.
>...

The LPD6803 does not work using any of those types, however its possible to make WLED work with those kind of LED strips.
I understand that the low level communication is provided by NeoPixelBus. 
Based on some already supported LED types by NeoPixelBus, I created two new files for the LPD6803 type:
Lpd6803ColorFeatures.h
Lpd6803GenericMethod.h

I have a 10 meters (2 x 5m) LED string working with WLED 0.11.0 and a Wemos D1 mini (ESP8266), for almost 10 days.
It works with GRB and BRG types, even when mixed (thanks to new enable color order override for a specific range of the strip,  on v0.11.0)

Already sent the files to Makuna. In case he decides to integrate them on NeoPixelsBus, will be easy add support for the LPD6803 to WLED, by doing minimal changes on NpbWrapper.h (possibly on the BusWrapper_h file on the dev version too).

thetweaker, 2020-12-18:
Confirmed!  It is working on a ESP8266
Used a older LED strip with **SC6803** IC 
Added the files to the neo lib  and changed the NpbWrapper.h
Files are @ [https://github.com/Makuna/NeoPixelBus/issues/388](https://github.com/Makuna/NeoPixelBus/issues/388)

WerkAG, 2020-12-19:
Thanks for having tried and confirmed that it is working.
I'm also using an ESP8266 but tried with an ESP32 too. My installation
is working for 12 consecutive days, and hope it continues working up to the end of Christmas season.
I saw that Makuna have marked the files as enhancement, so maybe he will include them on future releases. 
Bellow is the NpbWrapper.h file (WLED v 0.11.00) that I'm using, with the changes for the LPD6803 led strips.

[NpbWrapper.zip](https://github.com/Aircoookie/WLED/files/5718794/NpbWrapper.zip)

WerkAG, 2021-01-16:
> 
> 
> Confirmed! It is working on a ESP8266
> Used a older LED strip with **SC6803** IC
> Added the files to the neo lib and changed the NpbWrapper.h
> Files are @ [Makuna/NeoPixelBus#388](https://github.com/Makuna/NeoPixelBus/issues/388)

There are good news on this matter.
Makuna fixed bugs and greatly improved initial code for the LPD6803 strings. Probably the next official NeoPixelBus release will include support for those led strings. Then, maybe WLED will add support for them too.

Sousanator, 2021-01-23:
Thanks a lot for this!  I was able to follow this thread and get my "old" strips working with WLED 11.1
Feels good to have all my strips under one fully featured program now

kuno23, 2021-11-14:
> Thanks a lot for this! I was able to follow this thread and get my "old" strips working with WLED 11.1 Feels good to have all my strips under one fully featured program now

hi, is possible to post your bin file here, or sent me to me email. I have old strip with lpd6803/D705 driver and dont know how to drive it with wled. thanks

JeffreyKlassen, 2021-11-16:
I am also trying to figure out how to get this to work with the 6803 leds. I presume I need to compile a bin file but have not been able to figure that out. Would love to know if you guys are still running them and if you could provide the bin file you used.

blazoncek, 2021-11-16:
Just create a new entry in bus_wrapper.h and LED settings page and you are done.
If you are kind, you'll make PR so others can benefit too. 😉

JeffreyKlassen, 2021-11-16:
Haha if I can figure out how to do either of those things I will certainly share it :)  I am neither a coder or a regular user of github. I see kuno23 did a PR so hoping maybe he has an idea of how.
Thanks for your reply.

Sousanator, 2021-12-27:
I can see why this wasn't immediately done when Makuna updated NeoPixelsBus. npdwrapper.h was already depreciated in the v12 dev so adding the entry was a little more involved.

This is my first pull request so I'm hoping I did it correctly and we see this officially added in the next release

I'm including two compiled binaries if anyone wants to try it out.  One adds LPD6803 support on v0.13.0-b6, the other adds support for the dev branch pulled on December 27, 2021

[WLED-0.13.0-b6-LPD6803.zip](https://github.com/Aircoookie/WLED/files/7782029/WLED-0.13.0-b6-LPD6803.zip)
[WLED-0.13.0-b6-LPD6803_Dec-27-2021.zip](https://github.com/Aircoookie/WLED/files/7782030/WLED-0.13.0-b6-LPD6803_Dec-27-2021.zip)

closed 2022-08-16 by blazoncek

salopette, 2022-08-23:
Hello, how is the LPD6803 strip connected to the Wemos? I want to test it too!
Thanks

PR #2456 (2022-08-16): Added support for LPD6803
You can now select LPD6803 IC chip from the LED selection menu and use it.

Resolves #1500