wled/WLED #1501: LittleFS Error mounting (-84)

PDays, 2020-12-16:
Any thoughts as to why I am getting this error when trying to flash WLED to an ESP32 with 11.0? [19:28:38].pio\libdeps\esp32dev\LittleFS_esp32\src\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
[19:28:38]E (59) esp_littlefs: mount failed,  (-84)
[19:28:38]E (60) esp_littlefs: Failed to initialize LittleFS
[19:28:39]Ada

huggy-d1, 2020-12-16:
Refer to the wiki: https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary [ESP32 section]
Do you have a bootloader installed on the ESP32?  Are you loading the firmware to the proper memory address (see wiki)?
Where are these errors showing up?  How are you flashing?

Justdigit, 2020-12-18:
> Any thoughts as to why I am getting this error when trying to flash WLED to an ESP32 with 11.0? [19:28:38].pio\libdeps\esp32dev\LittleFS_esp32\src\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1} [...]

I have the same problem. 
https://github.com/Aircoookie/WLED/issues/1441

Bagunceiro, 2020-12-20:
I think it means you haven't formatted it. Take a look at the example "LittleFS_test" in the littleFS_esp32 library, and specifically the comment on line 5. In fact in my case all I needed to do was run this sketch and my problem (same as yours but in a different context) was solved.

djashjones, 2020-12-31:
I had this issue until I installed the bootloader then firmware as per the instructions 😎

nukeKid, 2021-01-08:
Have the same problem with this board :  https://www.amazon.de/dp/B071P98VTG/ref=twister_B07Z6CSD9K?_encoding=UTF8&psc=1

first i tried it with esphomeflasher than i install the bootloader for esp32 and the firmware as in the instructions with esptool.

i can boot into wled but my ws218b strip dont shines.

Justdigit, 2021-01-09:
Since I flash directly in visual code I don't see errors. I even see it flashes the partition.
Looks like esp home flasher can't handle it? Idk.

ejdprops, 2021-04-05:
I've tried all of this and *still* can't get the WLED binary to load without this littlefs error.  I've reached the point of insanity where I keep doing the same thing over and over expecting a different result

Sumtin, 2021-06-20:
I just installed v0.12 on an ESP32 for the first time on Windows 10.  Started with esphome-flasher, but got the same LittleFS error and was unable to install bootloader.  I had to switch over using esptool (v3) directly so I could control the memory space.  Made a small SOP for myself, seems to work fine for me:

1. Download ESPTool
   https://github.com/espressif/esptool/releases
   
2. Install Python 3.4+

3. Download **WLED_x.x.x_ESP32_with_bootloader.bin** and copy to same folder as esptool.exe
   https://github.com/Aircoookie/WLED/releases

4. Download and install USB-to-UART Drivers for Windows
   https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

5. Plug in NodeMCU into computer, in this example I've assigned COM10 port to UART drivers:
   Device Manager -> Ports (COM & LPT) -> Silicon Labs CP210x..... -> Properties -> Advanced -> COM Port Number -> COM10

6. In PowerShell: ./esptool -p COM10 erase_flash

7. In PowerShell: ./esptool -p COM10 -b 460800 write_flash 0x00000 WLED_x.x.x_ESP32_with_bootloader.bin

bluedke70, 2021-11-02:
How can you flash the ESP 32s with a Mac?

closed 2022-04-27 by stale[bot]

tschundler, 2022-08-19:
> LittleFS_test

Using the LittleFS_test with TWOPART undefined fixed it for me - seems it needed a reformat

aeonSolutions, 2023-05-04:
LittleFS does not allow a `"/"` basepath. when mounting 
 
Try changing to something else. For instance "/storage"  

Here's a code snippet that is working:

```
    if(true==fs.begin( false, "/storage", maxFiles , "storage")){ // !! Only allow one file to be open at a time instead of 10, saving 9x4 - 36KB of RAM          
      this->mserial->printStrln( "Drive init...done.");
    //this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
    //this->onBoardLED->statusLED(100, 1);
    }else if(fs.format()){
        this->mserial->printStrln("Drive formated sucessfully!");
      //  this->onBoardLED->led[0] = this->onBoardLED->LED_RED;
      //  this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
      //  this->onBoardLED->statusLED(100, 2);
        if(true==fs.begin( false, "/storage", maxFiles , "storage")){
          this->mserial->printStrln( "Drive init...done.");
       // this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
       // this->onBoardLED->statusLED(100, 1);
        }else{
          this->mserial->printStrln("Drive Mount Failed");
          return false;
        }   

    }else{
        this->mserial->printStrln("Drive init Failed");
     //   this->onBoardLED->led[0] = this->onBoardLED->LED_RED;
     //   this->mserial->printStrln("here 1");
     //   this->onBoardLED->statusLED(100, 5);
     //   this->mserial->printStrln("here 2");
        return false;
    }
```

The code above can be found here:
https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries