wled/WLED #1502: Can't sign into WLED AP WIFI

Michi1860, 2020-12-16:
After connecting successfully to the WLED AP with wled1234 there is no prompt to sign in. 
After typing http://4.3.2.1 into the webbrowser and waiting for a bit there is a network-timeout error.
What am i doing wrong?

huggy-d1, 2020-12-16:
You should wait a bit longer at the wifi connection page where you tell your note10pro to connect to the WLED-AP.  It will "capture" your phone and present the WLED interface where you can enter your local wifi credentials.  For 4.3.2.1, you cannot use a web browser.

sansillusion, 2020-12-16:
You may need to disable cell data to get your phone to connect to the IP,
some phone try to connect on the internet instead when connected to both an
wifi AP and cell data at the same time.

mike2nl, 2020-12-21:
XIAOMI NOTE 10 PRO has issues with that what @sansillusion posted before me.

Michi1860, 2020-12-21:
Don't really know what changed but it's working now... 
Thanks and close :)

closed 2020-12-22 by Aircoookie