wled/WLED #1503: How are you guys changing the wifi on a numbe of these?

l0c0luke, 2020-12-16:
I have 5 of these at my house. Love 'em. But I want to take them to my friends house and set up a scene. I won't know his wifi until I get over there. How is everybody mass updating wifi settings? If at all.

Or will I have to turn one on at a time, wait for it to go into AP mode, and redo each one individually?

Legsmaniac, 2020-12-16:
> Or will I have to turn one on at a time, wait for it to go into AP mode, and redo each one individually?

Yep. Unless you set up the WiFi in the code, recompile and flash each one.

huggy-d1, 2020-12-16:
You can do one, upload the firmware, then download the firmware to another.  I would try this at home to verify it uploads / saves the wifi settings during the firmware upload process.  I am not at all sure with the 0.11.x version and the little file system where the settings are stored.

closed 2021-06-16 by stale[bot]