wled/WLED #1504: WLED_AP doesn't show up (esp8266)

fufs, 2020-12-16:
So I tried using the official binary on my D1 mini, however, after flashing nothing happens. No AP showing up and the built-in LED blinks different patterns every time reboot the board. I did "erase_flash" twice and after reflashing the same behavior. I also noticed that the board gets really warm. I flashed Tasmota just to make sure that the board isn't dead and it worked without a problem. Can you make sure that the binaries are correct?

Aircoookie, 2020-12-16:
Hi, what is the exact name of the binary you used and what is your flasher?

fufs, 2020-12-16:
WLED_0.11.0_esp8266.bin and esptool with the command sudo python3 -m esptool write_flash 0x0 ./WLED_0.11.0_esp8266.bin.

Just flashed 0.10.2 with the same command (with the binary name changed of course) and am experiencing the same behavior.

fufs, 2020-12-16:
It's the same with Tasmotizer, although Tasmotizer is based on esptool, so nothing unexpected. 

I can add that the board is doing an inverted blink (On for 3s, off for ~300ms) and after a few of those blinks it stays on for a bit longer just to resume the blinking. The number of blinks is not the same every time.

Aircoookie, 2020-12-16:
That is the correct binary and command. I don't have an explanation why Tasmota works on the board and the WLED binary doesn't... Maybe someone else had similar problems before, but the access point not showing up is usually a binary/board mismatch which you surely don't have.

The LED pattern you are observing sounds like a bootloop, which shouldn't really happen without configuration changes. Do you have another board where you could check if it exhibits the same behavior?

fufs, 2020-12-16:
Unfortunately, I don't have another board to test it on. A quick note is that I hooked it up to a USB charger instead of my laptop and at the beginning, it was doing the same thing, but now it's blinking quite rapidly (still no AP). Might have to try to build a bin myself.

fufs, 2020-12-16:
Weird, build it on my own using platform.io and now it shows up normally. The only thing I changed is that I defined WLED_DEBUG to see whats wrong.

closed 2020-12-16 by fufs

fufs, 2020-12-16:
Well, I think that the module is bad. It sometimes works, sometimes doesn't. The voltage regulator is getting very hot, it keeps rebooting, etc. If anyone has the same issue, I hope this thread helps you.