wled/WLED #1506: First led is white, second is green and I have no control over them at all

EJ2Splashy, 2020-12-16:
Using ws2812b lights on an ESP8266 board, 60/m  300led 5m strip 5v

I’ve flashed the board and plugged it all in, wired it up correctly but when I turn the strip on the first led is white, the second was green and the rest were off but while writing this I unplugged the ground wire from the board and the green one turned yellow, and stayed yellow even after I put the ground wire back. I’m connected on WLED but I have no control. 
Please help if possible, thank you!

EJ2Splashy, 2020-12-16:
WLED version 0.11.0

huggy-d1, 2020-12-16:
Is your ESP8266 GND connected to the GND lead of the power supply AND the GND lead of the LED strip?  If it is not, please make that interconnection.

EJ2Splashy, 2020-12-18:
> Is your ESP8266 GND connected to the GND lead of the power supply AND the GND lead of the LED strip? If it is not, please make that interconnection.

Thanks for the reply, I’ll check when I’m home but I’m pretty sure, I’ve connected the led ground to the nodeMCU and the psu

WolfgangHusen, 2020-12-20:
same here: Using ws2812b lights on an ESP8266 board, 60/m 300led 5m strip 5v. 
Version 0.11.0 and 0.11.1.
everything is properly connected...

EJ2Splashy, 2020-12-20:
How many of ur leds turn on?

WolfgangHusen, 2020-12-20:
first try first was white, second green. after plugging and de-plugging... up to 12. now only first is green

EJ2Splashy, 2020-12-20:
Yeah that’s exactly what happens to me more turn on as u unplug and plug back in but I have no idea what the problem is

WolfgangHusen, 2020-12-20:
i give it a try now with a different led strip...

WolfgangHusen, 2020-12-20:
no change, different strip, first 1 green and 2 white; then 4 white... no control at all

WolfgangHusen, 2020-12-20:
UPDATE: I used another ESP8266 board - it works! And now the strange thing: I flashed onto the "buggy" board a 2801 binary, connected a 2801 strip, and it works, as well! (No surprise why the suizide rate of IT people is high...)

EJ2Splashy, 2020-12-20:
Wait so you just used a different board??

mike2nl, 2020-12-21:
@WolfgangHusen 
- May i ask what type of ESP8266 board you have used?
Clones are very bad, always believe me. They give you and others a lot of trouble. The clones are cheap>
Calculate once for yourself what the prices are for good hardware and bad hardware. You wiull drive a BMW but pay for a FIAT 500 only? ;-)

- Have you cleared the FLASH before burning a new fimrware?
This is sometimes 2x times needed depending on the quality of the Flash Memory.

This is what i do on Windows 7 and 10 machines when i help out others.
I make use of the Command (CMD) console in windows.

My esp tool is placed here: 
cd C:\Temp_Tools\esptool\0.4.13

And this is commend i use always twice to be sure. But between the two commands the esp board has to be disconneted and then reconnected and the send a second time the command. I had never bad flashes the last 4 years.

esptool -cp COM18 -cb 115200 -ce

The COM port is depending where you hardware is connected via USB.

t0mb3r, 2020-12-22:
I have the same problem, first led is green and cant connect. Issue is, I am testing it, and I have only one LED, so, basically, I dont know if its working or not, because only one 5mm led is connected. Stuck in green. Using a nodemcu, and 0.11.1

t0mb3r, 2020-12-22:
So i just tried a neopixel ring and it works, I actually need four 5mm neopixel led, would it work? Why when only one is connected it does not work?

WolfgangHusen, 2020-12-22:
@EJ2Splashy : yes, same type, just another one.
@mike2nl : 
- I am using these: https://www.amazon.de/gp/product/B074Q2WM1Y/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1; as I am working in automotive industry, I am always careful with those comparisons, but I know what you mean ;-)
- For flashing I used "ESPHome-Flasher-1.2.0-windows-x64". It comes along with a nice GUI. And if I am not wrong, it always starts with erasing old flash before it flashes the new

EJ2Splashy, 2020-12-23:
Ah thanks I’ll get some

closed 2021-06-16 by stale[bot]

blaskovicz, 2021-12-29:
Did anyone figure this out? Just swapping a board or reflashing is the solution? My LEDs were working fine up until tonight.

jkirkcaldy, 2023-10-17:
> Did anyone figure this out? Just swapping a board or reflashing is the solution? My LEDs were working fine up until tonight.

I had this problem, A sacrificial LED solved it for me.