wled/WLED #1507: LittleFS ESP8266

Bebick, 2020-12-16:
Hey,
i get this following fault of compiling the WLED Software in IDE:

In file included from C:\Users\PC\Downloads\WLED-master(6)\WLED-master\wled00\wled00.ino:13:0:
wled.h:55:24: fatal error: LittleFS.h: No such file or directory
   #include <LittleFS.h>
                        ^
compilation terminated.
exit status 1
LittleFS.h: No such file or directory

how can i solve it?

Bebick, 2020-12-16:
I am using the WLED 0.11 and the latest Arduino IDE to compile the code

huggy-d1, 2020-12-16:
Try using VS Code & Platform IO IDE (extension to VS Code).
Two compile / builds later and it's ready to install on your device.

keukenrol, 2020-12-18:
Same here, previous 10.2 compiled well. Libraries updated. Now it doesn't. Updating due to a bug when receiving E1.31 the device suddenly goes to a framerate of 1-2fps for 10-15 seconds and then recovering...

Multiple libraries found for "WiFi.h"
In file included from F:\Downloads\WLED\11\WLED-master\wled00\wled00.ino:13:0:
Used: C:\Users\timva\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
wled.h:68:24: fatal error: LITTLEFS.h: No such file or directory
Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
compilation terminated.
exit status 1
LITTLEFS.h: No such file or directory

huggy-d1, 2020-12-18:
Are you using Arduino IDE or VS Code & Platform IO IDE?  If you are using Arduino IDE, you have to manually manage all the libraries - which probably changed since 0.10.2.

Bebick, 2020-12-20:
who can read has a clear advantage... 
I following the instructions of these site https://github.com/Aircoookie/WLED/wiki/Compiling-WLED and now it works :D

closed 2020-12-20 by Bebick