wled/WLED #1508: H801 reboots when changing too quickly

Ikswoknib, 2020-12-16:
**Describe the bug**
H801 Controller reboots if changing Values to fast over MQTT.
I have some H801 Controllers flashed with WLED 0.11. I use an MQTT Server and am controlling it via Loxone.
If i change the colors or even the brightness too quickly the controller seems to reboot or even to lose wifi. It still does not do anything for about 10 seconds. For me it looks like it can't handle it if there come too many mqqt commands in too less time.
Loxone sends a lot of commands while swiping a slider for example, it does not only send the command when you release the slider, it sends the commands all the time while sliding, or even while using the color circle...

**To Reproduce**
Simply change the mqtt api values quite quick and often

**Expected behavior**
Should just work as it is.

**WLED version**
 - H801 Controller with ESP8266
 - Version 0.11.0
 - self-compiled  with 5CH Support and H801

Thank you for your help!

huggy-d1, 2020-12-16:
Do you want WLED source code changed so it can tolerate a denial of service attack from Loxone sliders and color circles?  Is it possible to configure your MQTT broker to throttle updating your H801 WLED 0.11.x devices?  Is it possible to configure Loxone so it stops kills your H801's?

I am not sure how WLED can ignore valid packets that come flooding in other than to crash if it gets abused.

Ikswoknib, 2020-12-17:
Hmm, I just tested today a little bit and i realised that the problem also occurs when i switch WLED Off with &T=0 command and after about 500ms switch it on again. Thats f.e. withe the command &T=1 &LX=099075095.
Then it reboots also. I think this should not be normal or should it ?

I am not sure if i can set Loxone to not to send the messages too quickliy, i can't find any option for that, but i'll check it. If i can set it up in the mqtt broker i am not sure, i'll check this too.

huggy-d1, 2020-12-17:
You may want to simply try adding a 1000µF capacitor across the 5V / GND input to the module to buffer power input.

We know WLED on ESP8266's is near the edge of processing capabilities.  Buffering the power (by adding a capacitor) will help for sudden power demand changes and reduce temporary brown-outs perceived by sensitive circuits in the ESP8266.

Ikswoknib, 2020-12-17:
thanks for that, i'll try this next few days

Ikswoknib, 2020-12-19:
I tried it with the capacitor but it does not work.
It is although when i turn it off and on again with only very small brightness, lets say 5%. So i think with this low amount of power, it should not be a problem of the 5V power source....
The Problem is also when i turn it off and on directly via the web page of wled. So it is not a problem with mqtt or something like that.

It is very strange, any ideas ?

Ikswoknib, 2021-01-07:
I tried using the Master Version, with this the Problem is gone... 
So i think there was something hanging in WLED Software...

closed 2021-06-18 by stale[bot]