wled/WLED #1509: Icon Font

gsieben, 2020-12-16:
A quick question about the icon font (wicon) that is coded into the "index.htm" file. Is this the whole font or just the used characters/icons? If there are other icons in it:Where can I look up which character/icon is the proper one for me? I ask because I have added an additional bottom table for "Relays". To be able to switch additional relays using alexa, mqtt, webinterface and the app.

Aircoookie, 2020-12-16:
Hi!
It is a custom font that contains just these characters to save space. Also see https://github.com/Aircoookie/WLED/issues/1246#issuecomment-706701045 :)

closed 2020-12-16 by gsieben

GrumpyMeow, 2020-12-19:
[Implemented svg->woff](https://github.com/Aircoookie/WLED/pull/1532)