wled/WLED #1510: Little Typo in msg.htm

gsieben, 2020-12-17:
**Describe the bug**
There ist a little type in the msg.htm file (and the automaitcally generated html_ui.h)

**To Reproduce**
Line 25: 			line-height: 200%%;
should be: 		line-height: 200%;

Cheers Gabriel

Aircoookie, 2020-12-17:
Hi!
Thank you, but this is not a bug/typo, you will see that it is correct when you open e.g. the `http://<WLED-IP>/teapot` page, which uses the `msg.htm`. The EspAsyncWebServer uses the % character as a delimiter for dynamically inserting content into the static HTML while sending it (similar to PHP), and `%%` will get converted to a single % in that process. If I would have just a single `%` it would get removed and the CSS would be invalid :)
Have a happy holiday season :)

closed 2020-12-17 by Aircoookie

gsieben, 2020-12-17:
I see. Many thank you for this really great project. I love it. It really saves me a lot of work. And my toilet floor looks just amazing now. 

have a nice vacation and holidays, too - stay healthy